RobotLocomotion/drake #16369: [multibody] Provide basic support for MuJoCo MJCF xml format

RussTedrake, 2022-01-13:
I've written a basic MJCF parser and am opening this issue to document my PR strategy and progress.  Reviewers will want to use the MuJoCo documentation, especially the XML scheme [here](https://mujoco.readthedocs.io/en/latest/XMLreference.html).

General strategy:
- I will incrementally add support for the essential aspects of the schema; the parser will log warnings for any unsupported tags, and throw errors if an unsupported tag would lead to erroneous behavior.
- The first PR will have almost no value... it will just establish the basic parser and add some test files.
- Once I support enough functionality to load all of the files in the OpenAI mujoco gym environment, then I will consider the parser to be potentially useful, and will add it to the main `Parser` class, including into `AddModelFromFile` and `AddModelFromString`, add python bindings, and close this issue.  We can add additional features/support incrementally thereafter.

Initial PR train:
- [ ] Establish basic parser and test files
- [ ] Parse world/anchored geometry using primitive shapes (no meshes)
- [ ] Parse inertial bodies
- [ ] Parse joints
- [ ] Parse motors

DamrongGuoy, 2022-01-13:
Does this mean, in some examples, we'll have side-by-side comparisons of MuJoco and Drake?

RussTedrake, 2022-01-14:
My goal here is to make models/environments/assets authored for MuJoCo available in Drake.  

If it also enables side-by-side comparisons, that is of course fantastic. But that will require running MuJoCo separately.  This issue has nothing to do with connecting to the mujoco source code nor library, only parsing the description format.

PR #17358 (2022-06-13): [multibody] Add support for MJCF to Parser.
This adds the plumbing to parse into the MuJoCo format (MJCF) from the
main Parser class.

Resolves #16369.

+@rpoyner-tri for feature review, please.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/17358)
commit 45ffba9d (2022-06-13): [multibody] Add support for MJCF to Parser. (#17358)

This adds the plumbing to parse into the MuJoCo format (MJCF) from the
main Parser class.

Resolves #16369.
closed 2022-06-13 by sammy-tri