RobotLocomotion/drake #16380: Add CI builds for Apple silicon (ARM64)

jwnimmer-tri, 2022-01-14:
Eventually, we would like to offer support for building and running Drake on Apple's newer hardware that uses ARM64, instead of x86_64.  To that end, we should have CI machines available and build names and scripting defined such that we can run experimental builds in Jenkins.

Relates to #16206, #14555, #13514.

BetsyMcPhail, 2022-02-03:
A ticket has been filed with MacStadium support to get some more information.

BetsyMcPhail, 2022-02-11:
VMWare, which we are currently using, does not currently have a projected timeline for support of the M1 platform.

The alternative on MacStadium is to switch to Orka 2.0, which is set to be released in early March. With Orka 2.0 we would be able to have a mix of M1 and Intel machines.

AWS is currently offering preview M1 instances but still has the 24 hour minimum allocation period.

BetsyMcPhail, 2022-02-22:
Update from our contact at Mac Stadium - hopefully we'll be getting to test out M1s a bit sooner than anticipated:  "Orka 2.0 has been released.  If it’s ok with you I’ll go ahead and enter a ticket to get you in line to get a free test cluster."

BetsyMcPhail, 2022-03-08:
We have an Orka 2.0 onboarding meeting with MacStadium on Thurs 3/10

BetsyMcPhail, 2022-03-10:
MacStadium needs some more time to properly configure the Orka installation, they will be rescheduling the meeting.

BetsyMcPhail, 2022-04-11:
Initial provisioned and unprovisioned images have been created in Orka.

Jenkins experimental jobs have also been created [here](https://drake-jenkins.csail.mit.edu/view/Mac%20M1/).

Jenkins is unable to SSH into the Orka VMs. The MacStadium networking team modified the site-to-site VPN with AWS to include Orka. We have an open ticket with them to help solve the SSH issue.

jwnimmer-tri, 2022-05-05:
Seems to be up and running, without any CI hitches!  Fixing the build is under #17009.

jwnimmer-tri, 2022-05-14:
Oops, I've overlooked something.  (Re-opening.)

We also need to have an "everything" variant of M1 builds, at least in Experimental:
- `mac-m1-monterey-unprovisioned-clang-bazel-experimental-everything-release`

If we have room for a nightly of the same, that would also be nice.  If we need to stick with only the one nightly for now, that's fine and we can keep the current one only.

BetsyMcPhail, 2022-05-16:
We do have room for a nightly build. I will add the -everything- nightly and experimental jobs.

BetsyMcPhail, 2022-05-17:
`mac-m1-monterey-unprovisioned-clang-bazel-(experimental|nightly)-everything-release` builds have been created.

I also moved the nightly M1 builds to non-production. They weren't in the Production tab on Jenkins yet, but were spamming failure emails every night.

jwnimmer-tri, 2022-07-13:
> Remaining work: debugging networking to the Gurobi license server.

Split out into #17565 for clarity.  Nothing else here remains, so closing.