RobotLocomotion/drake #16383: pip install's system prerequisites

jwnimmer-tri, 2022-01-14:
As it stands today (as of v0.38.0), Ubuntu 18 users must know to install `libpython3.6 python3-tk libx11-6 libsm6 libxt6 libglib2.0-0` by following our documentation at https://drake.mit.edu/pip.html.  Similarly for Ubuntu 20.  When they forget to do this, they get some obscure loader error.  Some people might not even read those docs.

To smooth that rough edge, we should make `import pydrake.all` fail with a good message in case these are missing, and also provide an easier path for users to install these.

At minimum, a text file with the list of packages to install for that version of Drake should be shipped as part of the wheel, rather the website documentation (we don't version the website).  The documentation (and runtime failure message) can refer to that file.  Having a bash script that does the extra install would be a nice bonus.

mwoehlke-kitware, 2022-01-14:
It's worth noting that, with the exception of Tk (which should already give a fairly reasonable error message), no one running outside of something like a Docker environment is likely to _not_ already have those packages...

jwnimmer-tri, 2022-01-14:
Yes, this is fairly low-priority in terms of the error message for users. 

The main win is for maintainability -- to de-brittle our unversioned website docs, and also de-duplicate the docs content vs wheel-test image code.

BetsyMcPhail, 2022-02-23:
@alesgenova this may a good issue to look at next

mwoehlke-kitware, 2022-09-29:
In the *wheel*, `pydrake/doc/INSTALLATION` seems like a reasonable place for this to live. Where should it live in a regular install? Where in the source tree?

jwnimmer-tri, 2022-09-29:
> In the wheel, pydrake/doc/INSTALLATION seems like a reasonable place for this to live.

Fine by me.

> Where should it live in a regular install?

Nowhere.  The file is pip-specific, and so should only exist in the `*.whl` binaries.  I wouldn't make sense for Docker or Apt binaries.  We could imagine having an analogous file for the Tgz binaries (with different instructions) but I don't think it's necessary.

> Where in the source tree?

Somewhere within `drake/tools/wheel/...`, wherever makes sense to you.

mwoehlke-kitware, 2022-09-29:
Um... that's going to be a little interesting; currently, the only things that go in the wheel that aren't in the Drake install are from drake-dependencies...

jwnimmer-tri, 2022-09-29:
I don't understand the difficulty you're alluding to.  At worst, it could be a here-doc in `build-wheel.sh`, but I assume that something like this could keep it as a separate file in `git` for convenience:

```patch
--- a/tools/wheel/Dockerfile
+++ b/tools/wheel/Dockerfile
@@ -64,6 +64,7 @@ RUN --mount=type=ssh \
     /image/build-drake.sh
 
 ADD image/build-wheel.sh /image/
+ADD image/INSTALLATION /image/
 ADD image/setup.py /opt/drake-wheel-build/wheel/
 
 ENV DRAKE_VERSION=${DRAKE_VERSION}
```

PR #18015 (2022-10-24): Improve error reporting when pydrake import fails
Modify pydrake (at least, `import pydrake.all`) to detect when it can't be loaded due to failure to load a dynamic library, and to give a somewhat more user-friendly error message. Add a brief document listing what distro packages need to be installed, which is referenced by the aforementioned error.

Fixes #16383.

- - -
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/18015)