RobotLocomotion/drake #16403: CI: Mac Debug jobs fail due to 'No space left on device'

BetsyMcPhail, 2022-01-19:
Currently, Big Sur Nightly Debug builds are failing:
https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-big-sur-clang-bazel-nightly-debug/
https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-big-sur-clang-bazel-nightly-everything-debug/

Monterey jobs are still passing but are at about 98% capacity when complete
https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-monterey-clang-bazel-nightly-debug/
https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-monterey-clang-bazel-nightly-everything-debug/

All other Mac jobs (e.g. all release) have plenty of disk space.

We have encountered similar issues in the past, see #13284 and #11587.

To try to see if any recent changes might have pushed the space over the edge, we ran a few experimental builds. For example: https://drake-jenkins.csail.mit.edu/view/Mac%20Big%20Sur/job/mac-big-sur-clang-bazel-experimental-debug/12/ uses SHA ffee3a5 and previously passed on CI but is now failing.

Disk space is a finite resource for Mac CI so we cannot just increase it without adjusting something else.

BetsyMcPhail, 2022-01-19:
Proposed plan:
- [ ] Audit existing images and see if there's any cleanup that can be done
- [ ] Move some Mac nightly production jobs to weekly production. There are currently 26 Mac nightly jobs and they run until at least mid-morning.
- [ ] Reduce the number of Mac jobs that may be launched at once (it's currently 8)
- [ ] Increase disk space on Macs

@jwnimmer-tri  does this sound like a reasonable plan? I'm happy to take other suggestions.

jwnimmer-tri, 2022-01-19:
> Audit existing images and see if there's any cleanup that can be done.

In particular, I would like to know:

(1) if the disk volume is the same (i.e., shared) for the OS stuff (homebrew files), versus the Drake build artifacts (typically, a cache folder in the user's home dir), and

(2) in a completed debug build, how much space is claimed by the OS files versus the Drake build artifacts.

In short, I'd like to understand profile where the space is going / what is the worst offender, so that we know where to optimize.

> (*)The current configuration is: (up to) 8 VMS x 160GB. The total needs to be about the same (or less).

I'm fine with the general plan of dividing up the total disk into larger per-VM slices, and thus needing to reduce the number of concurrent jobs.

I'd offer that typically the "Big Sur" vs "Monterey" distinction does not matter.  If it fails on one, it usually fails on both.  How about we keep the older one (Big Sur) as nightlies and bump the newer one ("Monterey") to weeklies?  I say that because we do our nightly and monthly releases from the older one, so it's more important in that sense.  We will still have coverage of Monterey via the Continuous builds (which are fairly quick).

BetsyMcPhail, 2022-01-19:
The plan for weeklies vs. nightlies sounds good to me.

BetsyMcPhail, 2022-01-19:
Mac Monterey Nightly jobs were moved to Weekly ([drake-jenkins-jobs script update](https://github.com/RobotLocomotion/drake-jenkins-jobs/commit/96965390c3b9a23e113c3bff04280e722f6f752c)) and the max number of VMs (Template Instance Cap in Jenkins) was reduced to 6. Let's see how long it takes everything to complete tonight to see if that's reasonable.

BetsyMcPhail, 2022-01-19:
Starting with an unprovisioned image, we installed the packages listed in the [drake-ci Brewfile](https://github.com/RobotLocomotion/drake-ci/blob/main/setup/mac/Brewfile) and ran the [install_prereqs script](https://github.com/RobotLocomotion/drake/blob/master/setup/mac/install_prereqs.sh). We also did a `brew cleanup` and looked for any other files that could be deleted. At this point, the images only contain what is specified by the setup scripts and necessary for CI. Whether this clean-up will give us enough diskspace is still TBD.

BetsyMcPhail, 2022-01-19:
To note, the OpenJDK runtime on _provisioned_ images was updated from 1.8.0_292 to 16.0.1.

Previously, both the unprovisioned and provisioned images had only **adoptopenjdk8** installed. This is the version required by Jenkins to launch nodes. After they are launched, the unprovisioned images then [install adoptopenjdk](https://github.com/RobotLocomotion/drake/blob/25d27c9ae5201f51e63cfe9b93526a4227c3e692/setup/mac/binary_distribution/Brewfile#L6).

On the new provisioned images,  both **adoptopenjdk8**  and **adoptopenjdk** are installed and Jenkins is configured to use the older version it requires. This way, both unprovisioned and provisioned images will be using the same set of packages as specified in the setup script.

#16404 has been opened because AdoptOpenJDK has been deprecated.

BetsyMcPhail, 2022-01-21:
The Monterey and Big Sur _provisioned_ images have been increased from 160GB to 200GB.

The max number of Mac jobs in Jenkins has been decreased to 4.

jwnimmer-tri, 2022-01-27:
This might be sufficiently improved to close this issue now?

BetsyMcPhail, 2022-01-27:
I agree, the nightly builds have been completing in the morning with no issues and continuous builds don't seem to be lagging.

ggould-tri, 2022-10-11:
Adding the phrase "OSError: [Errno 28]" here for search juice.  Fun fact, 28 has been the mac error code for disk space since the original mac in 1984, which didn't even have a hard disk!

BetsyMcPhail, 2022-10-12:
The  Big Sur provisioned images have been increased from 200GB to 228GB.

The max number of Mac jobs in Jenkins has been decreased from 8 to 7.

BetsyMcPhail, 2023-01-19:
These errors have appeared on Monterey nightly debug jobs, for example: https://drake-jenkins.csail.mit.edu/job/mac-x86-monterey-clang-bazel-nightly-everything-debug/67/

BetsyMcPhail, 2023-02-06:
The Monterey  x86 provisioned images have increased from 200Gi to 230Gi.

closed 2023-02-06 by BetsyMcPhail