RobotLocomotion/drake #16408: How to obtain access to the private RobotLocomotion/snopt GitHub repository?

littleggghost, 2022-01-20:
Hi there
I want to try with SNOPT, could I get your help?
https://drake.mit.edu/bazel.html#developing-drake-using-bazel

jwnimmer-tri, 2022-01-20:
Thanks for your interest!

The easiest way to try Drake along with SNOPT is to use one of our binary releases, rather than building from source.  The https://drake.mit.edu/installation.html page has the details, but it could be as simple as `pip install drake` within a python venv, or else downloading a nightly build, etc.  All of our binary releases incorporate a compiled version of SNOPT, free of charge.

https://drake.mit.edu/bazel.html#snopt says:
> The other mechanism is via access to a private RobotLocomotion git repository.

SNOPT is commercially licensed, so our RobotLocomotion private repository is for use only by the Drake team, and we cannot share access to it with others.  If you're associated with an academic institution, they might already have a license to use the SNOPT source code, or be able to buy one.  In that case you would use the `export SNOPT_PATH=...` when building from source.

jwnimmer-tri, 2022-01-22:
I'll close this now, but please feel free to comment if you have more questions.

closed 2022-01-22 by jwnimmer-tri