RobotLocomotion/drake #16422: SceneGraph should ignore collisions between two bodies welded to the same link

hongkai-dai, 2022-01-23:
Currently SceneGraph ignores collisions between two bodies both welded to the world. Following the same logic we should ignore the collision between two bodies both welded to the same link.

See slack discussion https://drakedevelopers.slack.com/archives/C2WBPQDB7/p1642886146051700

jwnimmer-tri, 2022-01-23:
See also #13672.

rpoyner-tri, 2022-02-01:
I think maybe the right place to land a fix is actually in MultibodyPlant, which is where the magic/implicit collision filtering is added now. Modifying the title to reflect this.

rpoyner-tri, 2022-04-06:
Also related: #11117.

rpoyner-tri, 2022-04-06:
I have some work-in-progress that (plausibly) does the right thing, but needs both some unit tests, and possibly more serious testing. @hongkai-dai , do you have any example code hanging around to help inspire some tests?

rpoyner-tri, 2022-04-08:
Nevermind. I have a test case going. The sad discovery is that the existing behavior (filter world-weld subgraph only) is completely untested. Still working on what to do about that.

PR #16920 (2022-04-11): [multibody] Filter collisions within welded subgraphs
Closes: #16422

This is a generalization of the the already-existing code to filter
collisions within bodies welded to the world.  Now, we filter collisions
within any welded subgraph of the model.

On the way to constructing a test, remove some unused infrastructure and
(mostly) prefer method calls to multiple optional constructor
parameters.

Also, back-fill a test that should have accompanied PR #1116.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/16920)
commit 9c662722 (2022-04-11): [multibody] Filter collisions within welded subgraphs (#16920)

* [multibody] Filter collisions within welded subgraphs

Closes: #16422

This is a generalization of the the already-existing code to filter
collisions within bodies welded to the world.  Now, we filter collisions
within any welded subgraph of the model.

On the way to constructing a test, remove some unused infrastructure and
(mostly) prefer method calls to multiple optional constructor
parameters.

Also, back-fill a test that should have accompanied PR #11116.
closed 2022-04-11 by rpoyner-tri