RobotLocomotion/drake #16433: Bring in libcmaes as a dependency.

hongkai-dai, 2022-01-25:
Now that [libcmaes](https://github.com/CMA-ES/libcmaes) has the Apach 2.0 license, I think it is ready to bring it as a Drake dependency

jwnimmer-tri, 2022-01-26:
@hongkai-dai will provide a small sample function (or program, or test) that calls cmaas, and then @jwnimmer-tri will open a PR with a sufficient BUILD file (+ releated) to get that working in Drake.

jwnimmer-tri, 2022-01-26:
A good sample program to use would be https://github.com/CMA-ES/libcmaes/blob/master/examples/sample-code.cc.

jwnimmer-tri, 2022-01-26:
@hongkai-dai for tracking purposes, could you document in this issue which Drake feature this dependency will be in service of?  Ideally, that new feature would have its own overall Drake tracking issue that you could just link to here.

jwnimmer-tri, 2023-09-23:
@hongkai-dai Do we have any timeline for using this CMA-ES library in Drake?  We added it 18 months ago as a speculative dependency ("dev" code) and haven't touched it since then (other than babysitting it to keep it up-to-date).

Now that #17231 is nearing completion, I either need to fix the CMA-ES to be hidden-safe or else remove it from Drake.  (We could bring it back later again if we want to, reverting the removal.)

hongkai-dai, 2023-09-23:
I ended up not using CMA-ES. I think we can close this issue.

PR #16441 (2022-01-27): [workspace] Add libcmaes as a testonly dependency
In the future, we might use this in production, but for getting started we only need it in the sample test.

Closes #16433.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/16441)
closed 2023-09-23 by hongkai-dai