RobotLocomotion/drake #16439: Redistribute mosek in Drake binary releases

RussTedrake, 2022-01-26:
Related to #10804, but it deserves it own issue for tracking.

Problem: Currently no solver that is available in the Drake binary release supports mixed-integer optimization, which in increasingly important.  Mosek is also a superior solver relative to many of the open-source solvers we make available for a number of optimization classes.

Proposed solution: Drake binary packages include the mosek binary; users would still have to provide the mosek license file.

This may involve an agreement w/ Mosek.

jwnimmer-tri, 2022-01-27:
The Mosek EULA section 8 discusses re-hosting.  I only spent a few moments reading it, but it seems like there is not a blanket grant for us to redistribute.  It does seem like we would need to get some kind of special permission.  Or possibly there is a redistribution grant somewhere else, that I've overlooked.

Or possibly we could have a drake "installer" that downloads and installs Mosek, adjacent to a Drake tarball.  We might not be able to add it to our Docker or Pip images, though.

The Mosek wheel (https://pypi.org/project/Mosek/#files) for `pip install Mosek` has the runtime library embedded in it:
```
  -rwxr-xr-x    324676  24-Jan-2022  09:09:38  mosek/libcilkrts.so.5
  -rwxr-xr-x  27536576  24-Jan-2022  09:09:38  mosek/libmosek64.so.9.3
```

For `pip install drake`, we might be able to reference that directly (instead of redistributing our own copy), as a dependency.

The question would be if it's okay (for our taste) for (say) `pip install drake==v0.39.0` to have an unconditional dependency on (say) `requirements: mosek >= 9.3, < 9.4`, or whether it would need to be somehow optional and opted-in at runtime.

In any case, if we could get permission to redistribute the Mosek runtime library (no need for headers) alongside Drake, that would be the easiest way forward.  We could include any notices or documentation that were important, as requested.

jwnimmer-tri, 2022-03-23:
@BetsyMcPhail -- Russ is working on the "permission" facet above, and will probably be finished soon.  I've added this to "On Deck" in anticipation.

After that's ready, it's over to us to work on the implementation here to make it happen.  The plan will be to turn on MOSEK during our Packaging builds (in addition to SNOPT).

I imagine it will be something like this:
- [ ] Ensure that when MOSEK is enabled in the build, that the proper files and notices are installed (if not already), and that any disclaimers we need on the website appear there now.
- [ ] Change all jobs in https://drake-jenkins.csail.mit.edu/view/Packaging/ to say "snopt-mosek-packaging" instead of just "snopt-packaging", and ensure that the `drake-ci` scripts notice this and correctly enable both SNOPT and MOSEK in the build and its test filters.
   - For the transition, we should probably to add Experimental builds for snopt-mosek-packaging while we test and develop things, without changing or removing any of the other builds.
- [ ] Change the build-wheel script to enable MOSEK and https://drake-jenkins.csail.mit.edu/view/Wheel/ to say "snopt-mosek-release" instead of just "snopt-release".

RussTedrake, 2022-04-06:
FYI.  We now have a fully-executed agreement with Mosek allowing us to redistribute the Mosek binaries with the Drake binaries.  Hoorah!  @BetsyMcPhail -- Let us please move forward on this.

jwnimmer-tri, 2022-04-11:
I'll sort this into the work queue in the appropriate place.  We need to switch to Focal (and related CI work) before we start this.

BetsyMcPhail, 2022-04-11:
While I was making changes to Jenkins, I added the following experimental jobs:

- https://drake-jenkins.csail.mit.edu/view/Packaging/job/linux-focal-unprovisioned-gcc-bazel-experimental-snopt-mosek-packaging/
- https://drake-jenkins.csail.mit.edu/view/Packaging/job/mac-big-sur-unprovisioned-clang-bazel-experimental-snopt-mosek-packaging/
- https://drake-jenkins.csail.mit.edu/view/Packaging/job/mac-monterey-unprovisioned-clang-bazel-experimental-snopt-mosek-packaging/

Nothing else has been updated but they are ready for testing when needed

BetsyMcPhail, 2022-04-11:
>  ensure that the drake-ci scripts notice this and correctly enable both SNOPT and MOSEK in the build and its test filters.

From the new [linux-focal-unprovisioned-gcc-bazel-experimental-snopt-mosek-packaging](https://drake-jenkins.csail.mit.edu/view/Packaging/job/linux-focal-unprovisioned-gcc-bazel-experimental-snopt-mosek-packaging/2/console) job:

`Bazel Command: CC=gcc CXX=g++ bazel  run --compilation_mode=opt --config=mosek --config=snopt --test_tag_filters=-gurobi //:install -- /opt/drake`

So no changes needed to the CI scripts.

BetsyMcPhail, 2022-04-12:
A PR to change all "snopt-packaging" jobs to "snopt-mosek-packaging" has been created. It is waiting and ready to coordinate with the other pieces of this issue.

jwnimmer-tri, 2022-04-12:
> I still need to publish the new license contract document before we can turn on MOSEK for the nightlies. I'm working on that.

Filed as => #16934.  Once that merges, we should be clear to land the PRs that turn on Mosek in our binaries.

BetsyMcPhail, 2022-04-12:
[Wheel job names in Jenkins ](https://drake-jenkins.csail.mit.edu/view/Wheel/) have also been updated in the same [drake-jenkins-jobs PR](https://github.com/RobotLocomotion/drake-jenkins-jobs/pull/3).

closed 2022-04-14 by jwnimmer-tri