RobotLocomotion/drake #16447: Upgrade to Bazel 5.0 in Provisioned

jwnimmer-tri, 2022-01-27:


jwnimmer-tri, 2022-01-27:
I have been running 5.0 locally since it was released, and I haven't seen any problems.  Our macOS unprovisioned is also running it without any problems.

FYI see #16405 for a fix (that's already landed) that is required for 5.0.  Users of older Drake releases that still want to use Bazel 5.0 should set `build --notrim_test_configuration` in the `drake/user.bazelrc` file to work around the problem.

jwnimmer-tri, 2022-01-31:
@BetsyMcPhail could you take this on (or delegate it), as part of this week's "monthly dependency upgrades" work?

(For myself, I will check that Anzu is also happy with Bazel 5.)

BetsyMcPhail, 2022-01-31:
> @BetsyMcPhail could you take this on (or delegate it), as part of this week's "monthly dependency upgrades" work?
> 

Sounds good

jwnimmer-tri, 2022-02-01:
Yes, perfect!

To be clear -- in this case, though, I think we will **not** need to update the minimum required version.  We have no evidence that bazel 4.0 is incompatible with current Drake.  If you'd like to locally install 4.0 and try a Release build to confirm or refute,  that's fine too, but is not especially important.

BetsyMcPhail, 2022-02-07:
Linux images have been updated.

BetsyMcPhail, 2022-02-07:
Mac images are also updated