RobotLocomotion/drake #16451: pointcloud_voxelization_test memcheck failure on Focal

mwoehlke-kitware, 2022-01-27:
There appears to be a memcheck failure on Focal (not present on Bionic?).

Raw output:

```
FAIL: //tools/workspace/voxelized_geometry_tools:pointcloud_voxelization_test (see $HOME/.cache/bazel/_bazel_$USER/51e544cc7eefa21145ddf6fca5239e3e/execroot/drake/bazel-out/k8-opt/testlogs/tools/workspace/voxelized_geometry_tools/pointcloud_voxelization_test/test.log)
INFO: From Testing //tools/workspace/voxelized_geometry_tools:pointcloud_voxelization_test:
==================== Test output for //tools/workspace/voxelized_geometry_tools:pointcloud_voxelization_test:
==20== Memcheck, a memory error detector
==20== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==20== Using Valgrind-3.15.0 and LibVEX; rerun with -h for copyright info
==20== Command: tools/workspace/voxelized_geometry_tools/pointcloud_voxelization_test
==20== 
[==========] Running 1 test from 1 test suite.
[----------] Global test environment set-up.
[----------] 1 test from PointCloudVoxelizationTest
[ RUN      ] PointCloudVoxelizationTest.Test
==21== 
==21== Warning: Can't execute setuid/setgid/setcap executable: /usr/bin/nvidia-modprobe
==21== Possible workaround: remove --trace-children=yes, if in effect
==21== 
==21== 
==21== HEAP SUMMARY:
==21==     in use at exit: 1,812,534 bytes in 35 blocks
==21==   total heap usage: 305 allocs, 270 frees, 3,668,740 bytes allocated
==21== 
==21== LEAK SUMMARY:
==21==    definitely lost: 0 bytes in 0 blocks
==21==    indirectly lost: 0 bytes in 0 blocks
==21==      possibly lost: 0 bytes in 0 blocks
==21==    still reachable: 1,812,534 bytes in 35 blocks
==21==         suppressed: 0 bytes in 0 blocks
==21== Reachable blocks (those to which a pointer was found) are not shown.
==21== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==21== 
==21== For lists of detected and suppressed errors, rerun with: -s
==21== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
Trying voxelizer [0] CPU/OpenMP
Raycasting time 0.00605908, filtering time 0.00111698, total time 0.00717605
Raycasting time 1.08341, filtering time 0.0020576, total time 1.08547
[       OK ] PointCloudVoxelizationTest.Test (1790 ms)
[----------] 1 test from PointCloudVoxelizationTest (1798 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test suite ran. (1850 ms total)
[  PASSED  ] 1 test.
==20== 
==20== HEAP SUMMARY:
==20==     in use at exit: 67,949 bytes in 11 blocks
==20==   total heap usage: 415 allocs, 404 frees, 3,737,041 bytes allocated
==20== 
==20== 56 bytes in 1 blocks are definitely lost in loss record 4 of 11
==20==    at 0x483DD99: calloc (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==20==    by 0x55CE6BA: ???
==20==    by 0x55DEE07: ???
==20==    by 0x555D5C7: ???
==20==    by 0x4867FD4: ??? (in /usr/local/cuda-11.5/targets/x86_64-linux/lib/libOpenCL.so.1.0.0)
==20==    by 0x4DE947E: __pthread_once_slow (pthread_once.c:116)
==20==    by 0x48698DE: clGetPlatformIDs (in /usr/local/cuda-11.5/targets/x86_64-linux/lib/libOpenCL.so.1.0.0)
==20==    by 0x123DCE: cl::Platform::get(std::vector<cl::Platform, std::allocator<cl::Platform> >*) (cl.hpp:2351)
==20==    by 0x127831: voxelized_geometry_tools::pointcloud_voxelization::opencl_helpers::GetAvailableDevices() (opencl_voxelization_helpers.cc:550)
==20==    by 0x11EDC7: voxelized_geometry_tools::pointcloud_voxelization::GetAvailableBackends() (pointcloud_voxelization.cpp:28)
==20==    by 0x11CBE9: voxelized_geometry_tools::(anonymous namespace)::PointCloudVoxelizationTest_Test_Test::TestBody() (pointcloud_voxelization_test.cpp:232)
==20==    by 0x1459E3: HandleSehExceptionsInMethodIfSupported<testing::Test, void> (gtest.cc:2607)
==20==    by 0x1459E3: void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) (gtest.cc:2643)
==20==    by 0x145C4D: Run (gtest.cc:2682)
==20==    by 0x145C4D: testing::Test::Run() (gtest.cc:2672)
==20==    by 0x145FA4: Run (gtest.cc:2861)
==20==    by 0x145FA4: testing::TestInfo::Run() (gtest.cc:2833)
==20==    by 0x146803: Run (gtest.cc:3015)
[... 155 lines cut ...]
   fun:clGetPlatformIDs
   fun:_ZN2cl8Platform3getEPSt6vectorIS0_SaIS0_EE
   fun:_ZN24voxelized_geometry_tools23pointcloud_voxelization14opencl_helpers19GetAvailableDevicesEv
   fun:_ZN24voxelized_geometry_tools23pointcloud_voxelization20GetAvailableBackendsEv
   fun:_ZN24voxelized_geometry_tools12_GLOBAL__N_136PointCloudVoxelizationTest_Test_Test8TestBodyEv
   fun:HandleSehExceptionsInMethodIfSupported<testing::Test, void>
   fun:_ZN7testing8internal35HandleExceptionsInMethodIfSupportedINS_4TestEvEET0_PT_MS4_FS3_vEPKc
   fun:Run
   fun:_ZN7testing4Test3RunEv
}
==20== LEAK SUMMARY:
==20==    definitely lost: 65,656 bytes in 4 blocks
==20==    indirectly lost: 640 bytes in 2 blocks
==20==      possibly lost: 0 bytes in 0 blocks
==20==    still reachable: 1,653 bytes in 5 blocks
==20==         suppressed: 0 bytes in 0 blocks
==20== Reachable blocks (those to which a pointer was found) are not shown.
==20== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==20== 
==20== For lists of detected and suppressed errors, rerun with: -s
==20== ERROR SUMMARY: 4 errors from 4 contexts (suppressed: 0 from 0)
================================================================================
```

Related to #14928.

jwnimmer-tri, 2022-01-27:
I am surprised to see `/usr/local/cuda-11.5` in the error message.  Is this message from CI, or locally?  Given the path, I suspect this is not from CI -- in which case, we should isolate CI's flavor of any errors like this, and then fix those.  Local user software is out-of-scope for Drake's valgrind support.

mwoehlke-kitware, 2022-01-27:
If it's only on "my" machine (note, this is not my *personal* machine, but our TRI workstation, which ought to be comparable to what other TRI developers are using), then I am not (presently) able to reproduce any other memcheck failures.

We should probably do new CI runs on the latest code (@BetsyMcPhail?). Hopefully there are not failures that are reproducible *only* on CI...

jwnimmer-tri, 2022-01-27:
> ... but our TRI workstation ...

Hmm, I have no idea why CUDA would have been installed on your TRI-specific workstation.  We should find whoever / however that happened, and undo it (and fix the process or documentation so that it doesn't happen again next time).

mwoehlke-kitware, 2022-01-28:
OpenCL appears to have the same problem? (Still only showing up on non-CI builds, though...)

jwnimmer-tri, 2022-01-28:
If I'm reading this correctly, then the relevant CI build here is:

https://drake-jenkins.csail.mit.edu/view/Focal%20Sanitizer%20and%20Valgrind/job/linux-focal-clang-bazel-experimental-valgrind-memcheck/

It's only complaint is about `iris_test` (with a timeout), not the `pointcloud_voxelization_test` mentioned here.

The `@voxelized_geometry_tools` external (and its unit test which is the subject of this ticket) is not in active use within Drake.  If CI is not complaining about memcheck on that unit test (because CI's OpenCL libraries are memcheck-clean, but your local OpenCL library is not), then we need not carry it any further.

closed 2022-01-28 by jwnimmer-tri