RobotLocomotion/drake #16459: iris_test times out on Focal

mwoehlke-kitware, 2022-01-28:
`iris_test` is (sometimes?) timing out on focal at least when run under memcheck.

Raw output:
```
[3:27:38 PM]  [11,403 / 11,410] 693 / 700 tests; Testing //geometry/optimization:iris_test (shard 3 of 4); 1322s linux-sandbox ... (2 actions, 1 running)
[3:30:45 PM]  
[3:30:45 PM]  TIMEOUT: //geometry/optimization:iris_test (Summary)
[3:30:45 PM]        /media/ephemeral0/ubuntu/workspace/linux-focal-clang-bazel-experimental-gurobi-valgrind-memcheck/_bazel_ubuntu/a7a0bf282bc0f54d52d99bfcbbf4ad53/execroot/drake/bazel-out/k8-dbg/testlogs/geometry/optimization/iris_test/shard_3_of_4/test.log
[3:30:45 PM]  INFO: From Testing //geometry/optimization:iris_test (shard 3 of 4):
[3:30:45 PM]  ==================== Test output for //geometry/optimization:iris_test (shard 3 of 4):
[3:30:45 PM]  ==22== Memcheck, a memory error detector
[3:30:45 PM]  ==22== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
[3:30:45 PM]  ==22== Using Valgrind-3.15.0 and LibVEX; rerun with -h for copyright info
[3:30:45 PM]  ==22== Command: geometry/optimization/iris_test
[3:30:45 PM]  ==22== 
[3:30:45 PM]  Using drake_cc_googletest_main.cc
[3:30:45 PM]  
[3:30:45 PM]  Note: This is test shard 3 of 4.
[3:30:45 PM]  [==========] Running 6 tests from 3 test suites.
[3:30:45 PM]  [----------] Global test environment set-up.
[3:30:45 PM]  [----------] 2 tests from IrisTest
[3:30:45 PM]  [ RUN      ] IrisTest.SmallBox
[3:30:45 PM]  [       OK ] IrisTest.SmallBox (5693 ms)
[3:30:45 PM]  [ RUN      ] IrisTest.BallInBoxNDims
[3:30:45 PM]  [       OK ] IrisTest.BallInBoxNDims (6605 ms)
[3:30:45 PM]  [----------] 2 tests from IrisTest (12308 ms total)
[3:30:45 PM]  
[3:30:45 PM]  [----------] 2 tests from SceneGraphTester
[3:30:45 PM]  [ RUN      ] SceneGraphTester.MakeHalfSpace
[3:30:45 PM]  [       OK ] SceneGraphTester.MakeHalfSpace (596 ms)
[3:30:45 PM]  [ RUN      ] SceneGraphTester.MultipleBoxes
[3:30:45 PM]  [       OK ] SceneGraphTester.MultipleBoxes (3213 ms)
[3:30:45 PM]  [----------] 2 tests from SceneGraphTester (3810 ms total)
[3:30:45 PM]  
[3:30:45 PM]  [----------] 2 tests from IrisInConfigurationSpaceTest
[3:30:45 PM]  [ RUN      ] IrisInConfigurationSpaceTest.SpheresPrismatic
[3:30:45 PM]  [       OK ] IrisInConfigurationSpaceTest.SpheresPrismatic (1009289 ms)
[3:30:45 PM]  [ RUN      ] IrisInConfigurationSpaceTest.ConvexConfigurationSpace
[3:30:45 PM]  -- Test timed out at 2022-01-27 20:30:29 UTC --
[3:30:45 PM]  ================================================================================
```
([complete log](https://drake-jenkins.csail.mit.edu/view/Focal%20Sanitizer%20and%20Valgrind/job/linux-focal-clang-bazel-experimental-gurobi-valgrind-memcheck/3/console))

jwnimmer-tri, 2022-01-28:
The timeout that we need to meet (1500 seconds within each test shard) is given here:

https://github.com/RobotLocomotion/drake/blob/504c6f182ccad3e01889a732e2550e3cb0c271eb/tools/dynamic_analysis/bazel.rc#L228

To diagnose this, I ran it locally in "one test case a time" mode, and scraped out the per-case timings:

```console
CC=clang-9 CXX=clang++-9 bazel test --compilation_mode=dbg --config=memcheck \
  //geometry/optimization:iris_test --test_output=streamed
...
[       OK ] DeprecatedIrisInConfigurationSpaceTest.JointLimits (1767 ms)
[       OK ] IrisInConfigurationSpaceTest.BlockOnGround (115332 ms)
[       OK ] IrisInConfigurationSpaceTest.BoxesPrismatic (21630 ms)
[       OK ] IrisInConfigurationSpaceTest.ConvexConfigurationSpace (651540 ms)
[       OK ] IrisInConfigurationSpaceTest.DoublePendulum (413723 ms)
[       OK ] IrisInConfigurationSpaceTest.JointLimits (235 ms)
[       OK ] IrisInConfigurationSpaceTest.SinglePendulum (152407 ms)
[       OK ] IrisInConfigurationSpaceTest.SpheresPrismatic (695647 ms)
[       OK ] IrisTest.BallInBox (1209 ms)
[       OK ] IrisTest.BallInBoxNDims (3693 ms)
[       OK ] IrisTest.ClosestPointFailure (8482 ms)
[       OK ] IrisTest.MultipleBoxes (18206 ms)
[       OK ] IrisTest.NoObstacles (1229 ms)
[       OK ] IrisTest.NoObstacles2 (362 ms)
[       OK ] IrisTest.SmallBox (15346 ms)
[       OK ] IrisTest.TerminationConditions (32674 ms)
[       OK ] SceneGraphTester.MakeBox (32 ms)
[       OK ] SceneGraphTester.MakeCapsule (39 ms)
[       OK ] SceneGraphTester.MakeCylinder (36 ms)
[       OK ] SceneGraphTester.MakeEllipsoid (28 ms)
[       OK ] SceneGraphTester.MakeHalfSpace (37 ms)
[       OK ] SceneGraphTester.MakeSphere (426 ms)
[       OK ] SceneGraphTester.MultipleBoxes (12270 ms)
```

In Jenkins the timings will vary somewhat due to the different hardware, but these numbers should be a good ballpark.

Rounding up to the nearest second, we have this set of timings:

```
[696, 652, 414, 153, 116, 33, 22, 19, 16, 13, 9, 4, 2, 2, 2, 1, 1, 1, 1, 1, 1, 1, 1]
```

To fit that list into grouping of at most 1500 seconds each, we need to place at most two tests cases into a single shard.  (The sum of the top 3 items is 1762.)

With 23 total test cases, we will need 12 shards in order to make that so.

Try this patch in CI and see if it resolves the problem:
```patch
--- a/geometry/optimization/BUILD.bazel
+++ b/geometry/optimization/BUILD.bazel
@@ -132,7 +132,7 @@ drake_cc_googletest(
 
 drake_cc_googletest(
     name = "iris_test",
-    shard_count = 4,
+    shard_count = 12,
     deps = [
         ":iris",
         "//common/test_utilities:eigen_matrix_compare",
```

jwnimmer-tri, 2022-01-28:
In the original problem report above, I missed that the failure log was coming from `gurobi-valgrind-memcheck`.  The data I posed above was in the default configuration, not `-gurobi`.

jwnimmer-tri, 2022-01-28:
I'll still keep this open to track the Mosek and/or Gurobi-specific fixes or work-arounds.

mwoehlke-kitware, 2022-02-09:
It's now timing out in ASAN builds also 😢 .

mwoehlke-kitware, 2022-02-11:
In light of #16567, maybe this can be closed?

PR #16461 (2022-01-28): iris_test: Run in 12 shards (#16459)
Increase `iris_test` shard count from 4 to 12; 4 is not enough to avoid timeouts under memcheck.

Fixes #16459.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/16461)
closed 2022-03-03 by jwnimmer-tri