RobotLocomotion/drake #16460: migrate non-standard URDF `capsule` tag to `drake:capsule`.

rpoyner-tri, 2022-01-28:
(Re-)discovrered while writing parsing docs. We support some non-standard legacy `capsule` tag in URDF, but the standard is unaware of it. Proposal: support `drake:capsule`, and maybe deprecate the made-up `capsule` tag. If URDF ever introduces `capsule`, at least we (hopefully by then) won't have old tags with potentially clashing syntax and semantics lying about.

FYI @EricCousineau-TRI @DamrongGuoy @joemasterjohn thoughts?

jwnimmer-tri, 2022-01-28:
Counter-proposal: add `<capsule>` to URDF using our current semantics, and change nothing in Drake.

jwnimmer-tri, 2022-01-28:
Here is the TODO in the current code:

https://github.com/RobotLocomotion/drake/blob/504c6f182ccad3e01889a732e2550e3cb0c271eb/multibody/parsing/detail_urdf_geometry.cc#L291-L304

It mentions https://github.com/ros/urdfdom_headers/pull/24.

rpoyner-tri, 2022-01-28:
I read the TODO; that's how I got here. The consequences of your counter-proposal are: (1) delete the TODO, (2) document some set of non-prefixed community extensions to URDF that Drake honors, (3) maybe delete that documentation if the standard version ever merges.

jwnimmer-tri, 2022-01-28:
Yes, that sounds correct to me.  I think our users are best served by keeping our current (quite reasonable) extension intact.

While the thoughts captured in the TODO make sense from a conformance perspective, I don't think they actually help users in any way.  If we had started with `<drake:capsule>` from day 1, perhaps there would be a case for keeping it, but rolling it back now doesn't help anyone.

rpoyner-tri, 2022-01-28:
Got it. I think I'll close the issue, and incorporate items 1 and 2 from above into #16444.

closed 2022-01-28 by rpoyner-tri