RobotLocomotion/drake #16494: tsan warning in RenderEngineVtk (Focal)

mwoehlke-kitware, 2022-02-02:
On Focal, several tests are failing with warnings related to `RenderEngineVtk`. These may have a single source.

Here is the raw output from `render_engine_vtk_minimal_example_test`:
```
[3:39:26 PM]  FAIL: //geometry/render/dev:render_engine_vtk_minimal_example_test (see /media/ephemeral0/ubuntu/workspace/linux-focal-clang-bazel-experimental-everything-thread-sanitizer/_bazel_ubuntu/8c089e08d90b0ce0a7c52d4e8bb78d87/execroot/drake/bazel-out/k8-dbg/testlogs/geometry/render/dev/render_engine_vtk_minimal_example_test/test.log)
[3:39:26 PM]  INFO: From Testing //geometry/render/dev:render_engine_vtk_minimal_example_test:
[3:39:26 PM]  ==================== Test output for //geometry/render/dev:render_engine_vtk_minimal_example_test:
[3:39:26 PM]  LLVMSymbolizer: error reading file: No such file or directory
[3:39:26 PM]  ==================
[3:39:26 PM]  WARNING: ThreadSanitizer: data race (pid=20138)
[3:39:26 PM]    Write of size 1 at 0x7fff225feb40 by main thread:
[3:39:26 PM]      #0 pthread_barrier_destroy <null> (render_engine_vtk_minimal_example_test+0x46c876)
[3:39:26 PM]      #1 <null> <null> (swrast_dri.so+0x148a56)
[3:39:26 PM]      #2 vtkNew<vtkRenderWindow>::~vtkNew() external/vtk/include/vtk-9.1/vtkNew.h:119:21 (render_engine_vtk_minimal_example_test+0x555229)
[3:39:26 PM]      #3 drake::geometry::render::RenderEngineVtk::RenderingPipeline::~RenderingPipeline() bazel-out/k8-dbg/bin/geometry/render/_virtual_includes/render_engine_vtk/drake/geometry/render/render_engine_vtk.h:204:10 (render_engine_vtk_minimal_example_test+0x55518c)
[3:39:26 PM]      #4 std::default_delete<drake::geometry::render::RenderEngineVtk::RenderingPipeline>::operator()(drake::geometry::render::RenderEngineVtk::RenderingPipeline*) const /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/unique_ptr.h:81:2 (render_engine_vtk_minimal_example_test+0x554f5e)
[3:39:26 PM]      #5 std::unique_ptr<drake::geometry::render::RenderEngineVtk::RenderingPipeline, std::default_delete<drake::geometry::render::RenderEngineVtk::RenderingPipeline> >::~unique_ptr() /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/unique_ptr.h:292:4 (render_engine_vtk_minimal_example_test+0x554eb7)
[3:39:26 PM]      #6 std::array<std::unique_ptr<drake::geometry::render::RenderEngineVtk::RenderingPipeline, std::default_delete<drake::geometry::render::RenderEngineVtk::RenderingPipeline> >, 3ul>::~array() /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/array:94:12 (render_engine_vtk_minimal_example_test+0x5546b8)
[3:39:26 PM]      #7 drake::geometry::render::RenderEngineVtk::~RenderEngineVtk() bazel-out/k8-dbg/bin/geometry/render/_virtual_includes/render_engine_vtk/drake/geometry/render/render_engine_vtk.h:92:7 (render_engine_vtk_minimal_example_test+0x55398f)
[3:39:26 PM]      #8 drake::geometry::render::RenderEngineVtk::~RenderEngineVtk() bazel-out/k8-dbg/bin/geometry/render/_virtual_includes/render_engine_vtk/drake/geometry/render/render_engine_vtk.h:92:7 (render_engine_vtk_minimal_example_test+0x5f34e9)
[3:39:26 PM]      #9 std::default_delete<drake::geometry::render::RenderEngine>::operator()(drake::geometry::render::RenderEngine*) const /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/unique_ptr.h:81:2 (render_engine_vtk_minimal_example_test+0x4fdf37)
[3:39:26 PM]      #10 std::unique_ptr<drake::geometry::render::RenderEngine, std::default_delete<drake::geometry::render::RenderEngine> >::~unique_ptr() /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/unique_ptr.h:292:4 (render_engine_vtk_minimal_example_test+0x4fc2b7)
[3:39:26 PM]      #11 std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, drake::copyable_unique_ptr<drake::geometry::render::RenderEngine> >::~pair() /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/stl_iterator.h:1286:12 (render_engine_vtk_minimal_example_test+0x1467bea)
[3:39:26 PM]      #12 void __gnu_cxx::new_allocator<std::__detail::_Hash_node<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, drake::copyable_unique_ptr<drake::geometry::render::RenderEngine> >, true> >::destroy<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, drake::copyable_unique_ptr<drake::geometry::render::RenderEngine> > >(std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, drake::copyable_unique_ptr<drake::geometry::render::RenderEngine> >*) /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/ext/new_allocator.h:153:10 (render_engine_vtk_minimal_example_test+0x1467bb9)
[3:39:26 PM]      #13 void std::allocator_traits<std::allocator<std::__detail::_Hash_node<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, drake::copyable_unique_ptr<drake::geometry::render::RenderEngine> >, true> > >::destroy<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, drake::copyable_unique_ptr<drake::geometry::render::RenderEngine> > >(std::allocator<std::__detail::_Hash_node<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, drake::copyable_unique_ptr<drake::geometry::render::RenderEngine> >, true> >&, std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, drake::copyable_unique_ptr<drake::geometry::render::RenderEngine> >*) /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/alloc_traits.h:497:8 (render_engine_vtk_minimal_example_test+0x1467b30)
[3:39:26 PM]      #14 std::__detail::_Hashtable_alloc<std::allocator<std::__detail::_Hash_node<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, drake::copyable_unique_ptr<drake::geometry::render::RenderEngine> >, true> > >::_M_deallocate_node(std::__detail::_Hash_node<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, drake::copyable_unique_ptr<drake::geometry::render::RenderEngine> >, true>*) /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/hashtable_policy.h:2102:7 (render_engine_vtk_minimal_example_test+0x1467ad6)
[3:39:26 PM]      #15 std::__detail::_Hashtable_alloc<std::allocator<std::__detail::_Hash_node<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, drake::copyable_unique_ptr<drake::geometry::render::RenderEngine> >, true> > >::_M_deallocate_nodes(std::__detail::_Hash_node<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, drake::copyable_unique_ptr<drake::geometry::render::RenderEngine> >, true>*) /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/hashtable_policy.h:2124:4 (render_engine_vtk_minimal_example_test+0x1467a75)
[3:39:26 PM]      #16 std::_Hashtable<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, drake::copyable_unique_ptr<drake::geometry::render::RenderEngine> >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, drake::copyable_unique_ptr<drake::geometry::render::RenderEngine> > >, std::__detail::_Select1st, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<true, false, true> >::clear() /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/hashtable.h:2028:13 (render_engine_vtk_minimal_example_test+0x1466ea5)
[3:39:26 PM]      #17 std::_Hashtable<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, drake::copyable_unique_ptr<drake::geometry::render::RenderEngine> >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, drake::copyable_unique_ptr<drake::geometry::render::RenderEngine> > >, std::__detail::_Select1st, std::equal_to<std::__cxx11::basic_string<char, std::char_tr [... cut ...]
[3:39:26 PM]  SUMMARY: ThreadSanitizer: data race (/media/ephemeral0/ubuntu/workspace/linux-focal-clang-bazel-experimental-everything-thread-sanitizer/_bazel_ubuntu/8c089e08d90b0ce0a7c52d4e8bb78d87/execroot/drake/bazel-out/k8-dbg/bin/geometry/render/dev/render_engine_vtk_minimal_example_test+0x46c876) in pthread_barrier_destroy
[3:39:26 PM]  ==================
[3:39:26 PM]  ThreadSanitizer: reported 1 warnings
[3:39:26 PM]  ================================================================================
```

([Full console output](https://drake-jenkins.csail.mit.edu/view/Focal%20Sanitizer%20and%20Valgrind/job/linux-focal-clang-bazel-experimental-everything-thread-sanitizer/4/consoleFull))

jwnimmer-tri, 2022-02-02:
Note the path `//geometry/render/dev`.  For dev code, our response should be to disable the test in that configuration `tags = no_tsan`.

jwnimmer-tri, 2022-02-02:
Ah, in the full log we also have `//geometry/render:render_engine_vtk_test` and similar failures.  That's more relevant.

jwnimmer-tri, 2022-02-02:
This error message is almost certainly a false positive from the mesa library.  If you can try installing `libgl1-mesa-dri-dbgsym` from `deb http://ddebs.ubuntu.com focal-updates main restricted universe multiverse` sources list, we might get more information.  In any case, our TSan suppression file will probably need to gain an entry to disable this report.

mwoehlke-kitware, 2022-02-02:
Right; I picked `render_engine_vtk_minimal_example_test` somewhat arbitrarily (it was first, and some of the other reports are longer); there are similarities in the traces that make it look like there may well be a single issue appearing in multiple places. (Also why I'm not opening multiple issues yet.)

FTR, the tests with warnings are:
- `//examples/manipulation_station:manipulation_station_test`
- `//examples/manipulation_station:proof_of_life_test`
- `//examples/scene_graph:bouncing_ball_run_dynamics_test`
- `//geometry/render:render_engine_vtk_test`
- `//geometry/render/dev:render_engine_vtk_minimal_example_test`

mwoehlke-kitware, 2022-02-02:
> This error message is almost certainly a false positive from the mesa library.

I was wondering about that also.

> If you can try installing `libgl1-mesa-dri-dbgsym` from `deb http://ddebs.ubuntu.com focal-updates main restricted universe multiverse` sources list ([1]), we might get more information.

@BetsyMcPhail ?

mwoehlke-kitware, 2022-02-02:
BTW, installing that locally appears to have had no effect; I still get:
```
    #0 pthread_create <null> (bouncing_ball_run_dynamics_test+0x4541fb)
    #1 <null> <null> (swrast_dri.so+0x147cc6)
    #2 drake::geometry::render::RenderEngineVtk::DoRenderDepthImage(drake::geometry::render::DepthRenderCamera const&, drake::systems::sensors::Image<(drake::systems::sensors::PixelType)6>*) const geometry/render/render_engine_vtk.cc:241:3 (bouncing_ball_run_dynamics_test+0x57d204)
...
```

jwnimmer-tri, 2022-02-02:
Possibly the symbolizer doesn't find these paths by default?
```
/usr/lib/debug/.build-id/6e/6af4b6e3ae935b52d94709f5f736ee178cd655.debug
/usr/lib/debug/.build-id/c9/0f484520f5f539fe8db1161e214eceb3264fbf.debug
/usr/lib/debug/.dwz/x86_64-linux-gnu/libgl1-mesa-dri.debug
```

mwoehlke-kitware, 2022-02-02:
...maybe not? I have no idea how `llvm-symbolizer` works. I *can* see that this:
```
llvm-symbolizer-9 --obj=bazel-bin/geometry/render/render_engine_vtk_test 0x528267
```
...produces something "useful", while this:
```
llvm-symbolizer-9 --obj=/usr/lib/x86_64-linux-gnu/dri/swrast_dri.so 0x147cc6
```
...does not.

jwnimmer-tri, 2022-10-12:
I'm OK leaving the suppression in place indefinitely.

closed 2022-10-12 by jwnimmer-tri