RobotLocomotion/drake #16500: Enable Fission (split debug symbols)

jwnimmer-tri, 2022-02-03:
If we add the below stanza to a bazel's settings, builds with debug symbols `-g` will probably be much smaller on disk.

```
# Enable fission to make -c dbg builds much smaller on disk.
build --fission=dbg
build --features=per_object_debug_info
```

We use this setting in Anzu to good effect.

For Drake, we'll need to make sure that it works well with all of our different tests, configurations, and extra CI.

For SNOPT builds, we'll also need to ensure that debug symbols are unavailable for SNOPT by default.  (We currently do this by checking / stripping the binaries of symbols; if symbols move to a different file, we might have opened a loophole.)

Depending on how our sanitizer (etc.) builds are set up, we might want `build --fission=all` instead of only `dbg`.

We should enable this first only for Ubuntu as the priority (for pre-merge & local developer ease), and then later see if we can do the same for macOS.

mwoehlke-kitware, 2022-02-09:
```
ERROR: While parsing option --fission=all: Not a valid compilation mode: 'all' (should be fastbuild, dbg or opt)
```

I suppose you meant `--fission=yes` (which confusingly is not mentioned in the error message)?

jwnimmer-tri, 2022-02-09:
Oops, yes.  I meant "yes".

https://docs.bazel.build/versions/main/command-line-reference.html#flag--fission

mwoehlke-kitware, 2022-02-14:
Note: AFAICT, this is going to have the effect of removing debug symbols from some installs. Do we want to push this through regardless and tackle that later, or hold off until that can be addressed?

jwnimmer-tri, 2022-02-14:
None of our stable releases ship any debug symbols (they are all Release builds), so that aspect won't be a blocker.

Sometimes users will run a CMake Debug build to perform a local install, in order to debug some problem they are having.  One easy solution there might be to disable fission when invoked via CMake, with some configuration tweaks (adding `--fission=no`) near here:

https://github.com/RobotLocomotion/drake/blob/a2053d987464db0ce1d75b030001670ceabab9a6/CMakeLists.txt#L479-L494

mwoehlke-kitware, 2022-02-15:
BTW, local testing suggests this may increase build memory usage by ~7GiB.

jwnimmer-tri, 2022-03-14:
Closed by #16587.

closed 2022-03-14 by jwnimmer-tri

BetsyMcPhail, 2022-03-14:
I may have moved this into the "Done" column prematurely. This was completed for Ubuntu, I don't remember if we also wanted to try on Mac?

jwnimmer-tri, 2022-03-14:
Good point.  Could you open a new ticket for doing this on macOS?  I think it'll be easier to track that way.

The reason to do this on macOS would be to reduce disk space used by CI.  To that end, the first task in the new ticket would be to measure the disk footprint of a full debug build with an without fission.  If it's a useful saving, then we can pursue trying to roll it out.