RobotLocomotion/drake #16524: refactor RigidTransform IsIdentityToEpsilon => IsNearlyIdentity?

RussTedrake, 2022-02-05:
Currently `RigidTransform` has 
- `IsExactlyEqualTo`
- `IsNearlyEqualTo`
- `IsExactlyIdentity`, and...
- `IsIdentityToEpsilon` 

I find myself often typing `IsNearlyIdentity` and getting reminded by the compiler.  Can we pick one and stick?

mitiguy, 2022-02-07:
@RussTedrake   I am starting work on this now.

jwnimmer-tri, 2022-02-08:
For the record, for some future `RotationMatrix` function renaming, I'd argue that the `-ToInternalTolerance` suffix is not helpful.  The function takes no arguments, obviously the tolerance is built-in.  Adding that detail to the function name just makes the call sites unreadable.

If we ever wanted to offer a the tolerance as argument in the future, we'd want to do `boolean<T> IsNearlyIdentity(double tolerance = get_internal_tolerance_for_orthonormality()) const;` without needing to do rename the function yet again.

mitiguy, 2022-02-08:
Thanks @jwnimmer-tri  
Unless I hear otherwise from @RussTedrake, I'll proceed with that in a subsequent PR.

New method:
boolean<T> RotationMatrix::IsNearlyIdentity(double tolerance = get_internal_tolerance_for_orthonormality()) const;     
will replace old method:
boolean<T> RotationMatrix::IsIdentityToInternalTolerance();

PR #16541 (2022-02-09): Rename RigidTransform::IsIdentityToEpsilon to IsNearlyIdentity
Closes #16524.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/16541)
closed 2022-02-09 by mitiguy