RobotLocomotion/drake #16551: refactor RotationMatrix::IsIdentityToInternalTolerance() => IsNearlyIdentity()

mitiguy, 2022-02-09:
This follows issue #16524 that was resolved with PR #16541 which changed:

RigidTransform::IsIdentityToEpsilon(double translation_tolerance) to
RigidTransform::IsNearlyIdentity(double translation_tolerance);

Similarly, the related methods in RotationMatrix are:
Public methods:
RotationMatrix::IsExactlyIdentity();
RotationMatrix::IsIdentityToInternalTolerance();
RotationMatrix::IsNearlyEqualTo(const RotationMatrix& other, double tolerance);
RotationMatrix::IsExactlyEqualTo(const RotationMatrix& other);
Private method:
static boolean IsNearlyEqualTo(const Matrix3& R, const Matrix3& other, double tolerance);

For consistency, and per Jeremy's suggestion in issue #16524, change
RotationMatrix::IsIdentityToInternalTolerance() to
RotationMatrix::IsNearlyIdentity(double tolerance = get_internal_tolerance_for_orthonormality()) const;

xuchenhan-tri, 2022-05-04:
Looks like `RotationMatrix::IsIdentityToInternalTolerance()` is already deprecated. Are we ready to close this issue?

mitiguy, 2022-05-16:
Resolved with PR #16541.

closed 2022-05-16 by mitiguy