RobotLocomotion/drake #16580: LinearConstraint should store a sparse matrix

hongkai-dai, 2022-02-12:
I have a Drake MathematicalProgram object that takes 60GB of memory to construct.

One of the problem is that in LinearConstraint, it stores a dense matrix `A` for lb <= A*x <= ub https://github.com/RobotLocomotion/drake/blob/4d11342274394ad2d90f7cfdb12dc088b5bd55d0/solvers/constraint.h#L624. But in many cases `A` is a sparse matrix. For example, when we construct a BoundingBoxConstraint, which is derived from LinearConstraint, we set its `A` matrix to the dense identity matrix. This takes up a lot of memory.

I propose we change the internal storage type of `A` to Eigen::SparseMatrix<double>.

cc @jwnimmer-tri

jwnimmer-tri, 2022-02-12:
Would it make sense to change the storage for BoundingBoxConstraint, but leave LinearConstraint (and LinearEqualityConstraint) alone?  How sparse or dense is the typical non-bbox LinearConstraint?

jwnimmer-tri, 2022-02-12:
My first bid would be to remove the implementation inheritance, and thus there is no `A_` in the first place.

hongkai-dai, 2022-04-28:
Close this in lieu of #16894

closed 2022-04-28 by hongkai-dai