RobotLocomotion/drake #16589: this trivial linear program causes a core dump

RussTedrake, 2022-02-15:
```
from pydrake.solvers.mathematicalprogram import MathematicalProgram, Solve

prog = MathematicalProgram()
J = prog.NewContinuousVariables(2, "J")

prog.AddLinearCost(J[0] + J[1])

result = Solve(prog)
```
both on deepnote and my local machine i get a core dump.
```
russt@RussPuget:~/underactuated$ python3 test.py
free(): invalid next size (fast)
Aborted (core dumped)
```

RussTedrake, 2022-02-15:
ChooseBestSolver returns CLP.  I'll see if I can reproduce it in C++.

RussTedrake, 2022-02-15:
Adding this test to `clp_solver_test.cc` also reproduces the error:
```
GTEST_TEST(ClpSolverTest, TestIssue16589) {
  MathematicalProgram prog;
  auto J = prog.NewContinuousVariables(2, "J");

  prog.AddLinearCost(J[0] + J[1]);

  ClpSolver solver;
  // The program is infeasible.
  if (solver.available()) {
    auto result = solver.Solve(prog, {}, {});
    EXPECT_EQ(result.get_solution_result(),
              SolutionResult::kUnbounded);
  }
}
```
and running under valgrid reports
```
[ RUN      ] ClpSolverTest.TestIssue16589
==20== Invalid write of size 8
==20==    at 0x4F4EE72: ClpModel::emptyProblem(int*, double*, bool) (in /usr/lib/x86_64-linux-gnu/libClp.so.1.14.5)
==20==    by 0x4FCDFA0: ClpSimplex::startup(int, int) (in /usr/lib/x86_64-linux-gnu/libClp.so.1.14.5)
==20==    by 0x501DAF9: ClpSimplexPrimal::primal(int, int) (in /usr/lib/x86_64-linux-gnu/libClp.so.1.14.5)
==20==    by 0x4FCEE19: ClpSimplex::primal(int, int) (in /usr/lib/x86_64-linux-gnu/libClp.so.1.14.5)
==20==    by 0x1C4311: drake::solvers::ClpSolver::DoSolve(drake::solvers::MathematicalProgram const&, Eigen::Matrix<double, -1, 1, 0, -1, 1> const&, drake::solvers::SolverOptions const&, drake::solvers::MathematicalProgramResult*) const (clp_solver.cc:361)
==20==    by 0x1D9339: drake::solvers::SolverBase::Solve(drake::solvers::MathematicalProgram const&, std::optional<Eigen::Matrix<double, -1, 1, 0, -1, 1> > const&, std::optional<drake::solvers::SolverOptions> const&, drake::solvers::MathematicalProgramResult*) const (solver_base.cc:78)
==20==    by 0x1D886E: drake::solvers::SolverBase::Solve(drake::solvers::MathematicalProgram const&, std::optional<Eigen::Matrix<double, -1, 1, 0, -1, 1> > const&, std::optional<drake::solvers::SolverOptions> const&) const (solver_base.cc:33)
==20==    by 0x163CF6: drake::solvers::test::ClpSolverTest_TestIssue16589_Test::TestBody() (clp_solver_test.cc:226)
==20==    by 0x2530F3: HandleSehExceptionsInMethodIfSupported<testing::Test, void> (gtest.cc:2607)
==20==    by 0x2530F3: void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) (gtest.cc:2643)
==20==    by 0x25335D: Run (gtest.cc:2682)
==20==    by 0x25335D: testing::Test::Run() (gtest.cc:2672)
==20==    by 0x2536B4: Run (gtest.cc:2861)
==20==    by 0x2536B4: testing::TestInfo::Run() (gtest.cc:2833)
==20==    by 0x253F13: Run (gtest.cc:3015)
==20==    by 0x253F13: testing::TestSuite::Run() (gtest.cc:2993)
==20==    by 0x254649: testing::internal::UnitTestImpl::RunAllTests() (gtest.cc:5855)
==20==    by 0x253903: HandleSehExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool> (gtest.cc:2607)
==20==    by 0x253903: bool testing::internal::HandleExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) (gtest.cc:2643)
==20==    by 0x253B19: testing::UnitTest::Run() (gtest.cc:5438)
==20==    by 0x14931E: RUN_ALL_TESTS (gtest.h:2491)
==20==    by 0x14931E: main (drake_cc_googletest_main.cc:28)
==20==  Address 0x6b90e28 is 8 bytes before a block of size 16 alloc'd
==20==    at 0x483C583: operator new[](unsigned long) (vg_replace_malloc.c:433)
==20==    by 0x4F4EE52: ClpModel::emptyProblem(int*, double*, bool) (in /usr/lib/x86_64-linux-gnu/libClp.so.1.14.5)
==20==    by 0x4FCDFA0: ClpSimplex::startup(int, int) (in /usr/lib/x86_64-linux-gnu/libClp.so.1.14.5)
==20==    by 0x501DAF9: ClpSimplexPrimal::primal(int, int) (in /usr/lib/x86_64-linux-gnu/libClp.so.1.14.5)
==20==    by 0x4FCEE19: ClpSimplex::primal(int, int) (in /usr/lib/x86_64-linux-gnu/libClp.so.1.14.5)
==20==    by 0x1C4311: drake::solvers::ClpSolver::DoSolve(drake::solvers::MathematicalProgram const&, Eigen::Matrix<double, -1, 1, 0, -1, 1> const&, drake::solvers::SolverOptions const&, drake::solvers::MathematicalProgramResult*) const (clp_solver.cc:361)
==20==    by 0x1D9339: drake::solvers::SolverBase::Solve(drake::solvers::MathematicalProgram const&, std::optional<Eigen::Matrix<double, -1, 1, 0, -1, 1> > const&, std::optional<drake::solvers::SolverOptions> const&, drake::solvers::MathematicalProgramResult*) const (solver_base.cc:78)
==20==    by 0x1D886E: drake::solvers::SolverBase::Solve(drake::solvers::MathematicalProgram const&, std::optional<Eigen::Matrix<double, -1, 1, 0, -1, 1> > const&, std::optional<drake::solvers::SolverOptions> const&) const (solver_base.cc:33)
==20==    by 0x163CF6: drake::solvers::test::ClpSolverTest_TestIssue16589_Test::TestBody() (clp_solver_test.cc:226)
==20==    by 0x2530F3: HandleSehExceptionsInMethodIfSupported<testing::Test, void> (gtest.cc:2607)
==20==    by 0x2530F3: void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) (gtest.cc:2643)
==20==    by 0x25335D: Run (gtest.cc:2682)
==20==    by 0x25335D: testing::Test::Run() (gtest.cc:2672)
==20==    by 0x2536B4: Run (gtest.cc:2861)
==20==    by 0x2536B4: testing::TestInfo::Run() (gtest.cc:2833)
==20==    by 0x253F13: Run (gtest.cc:3015)
==20==    by 0x253F13: testing::TestSuite::Run() (gtest.cc:2993)
==20==    by 0x254649: testing::internal::UnitTestImpl::RunAllTests() (gtest.cc:5855)
==20==    by 0x253903: HandleSehExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool> (gtest.cc:2607)
[... cut ...]
   fun:_ZN7testing9TestSuite3RunEv
   fun:_ZN7testing8internal12UnitTestImpl11RunAllTestsEv
   fun:HandleSehExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>
   fun:_ZN7testing8internal35HandleExceptionsInMethodIfSupportedINS0_12UnitTestImplEbEET0_PT_MS4_FS3_vEPKc
   fun:_ZN7testing8UnitTest3RunEv
   fun:RUN_ALL_TESTS
   fun:main
}
[       OK ] ClpSolverTest.TestIssue16589 (11 ms)
```
That makes me think the problem is inside CLP, but I'll look some more when I'm back at the keys later.

RussTedrake, 2022-02-15:
I believe the problem is [here](https://github.com/RobotLocomotion/drake/blob/d2b3c1021949e4db8d948fcf577a40acc1622f0e/solvers/clp_solver.cc#L31-L36): If we do not have any constraints, then we are passing null pointers into Clp.

@hongkai-dai -- can you take a look after your paper deadline?

hongkai-dai, 2022-02-15:
Thanks for the investigation. I will fix it tomorrow.

hongkai-dai, 2022-02-15:
I am not completely sure if the null pointers (constraint_mat.innerIndexPtr(), constraint_mat.valuePtr(), constraint_lower.data(), constraint_upper.data()) in the block https://github.com/RobotLocomotion/drake/blob/d2b3c1021949e4db8d948fcf577a40acc1622f0e/solvers/clp_solver.cc#L31-L36 are the culprit here.

We also test CLP on a [QP](https://github.com/RobotLocomotion/drake/blob/02df22aa80659658ef5901fb0bf9556297a710a3/solvers/test/clp_solver_test.cc#L71-L110) with no constraint . In this QP these 4 pointers are also nullptr, but the code runs fine.

I am inclined to believe this is a problem in CLP upstream code. I will try to reproduce the error with CLP native code without using Drake.

cohnt, 2025-05-07:
FYI I don't get a core dump nowadays. Maybe the issue fixed itself?
```
from pydrake.all import MathematicalProgram, ClpSolver

prog = MathematicalProgram()
J = prog.NewContinuousVariables(2, "J")
prog.AddLinearCost(J[0] + J[1])

solver = ClpSolver()
result = solver.Solve(prog)
```