RobotLocomotion/drake #16615: Report the type of contact for the whole system.

hongkai-dai, 2022-02-17:
I often find that I need to check the type of contact for each pair of collision geometries, as we support different types under different situations. Currently we have

1. compliant-to-compliant hydroelastic contact.
2. rigid-to-compliant hydroelastic contact.
3. point-to-point contact.

And depending on the scalar type (T = double versus T=autodifff), the same pair of geometries can incur different contact.

Hence I often need to write a function to loop through `SceneGraphInspector.GetCollisionCandidates()`, and print out the name of the collision geometry. Now I also need to print out the properties (rigid / compliant) of that geometry, and the type of contact.

Would it make sense to provide a function inside Drake to print these information?

DamrongGuoy, 2022-02-22:
This is a relevant slack [conversation](https://drakedevelopers.slack.com/archives/C2WBPQDB7/p1645558851730439) that might be handy later.  It's about looking for all collision geometries of type compliant_hydroelastic.

This is @joemasterjohn's code that could be useful:
```C++
  GeometrySet all_geometries;
  for (auto body : plant.GetBodyIndices(model)) {
    for (auto collision :
         plant.GetCollisionGeometriesForBody(plant.get_body(body))) {
      if (scene_graph.GetProximityProperties(collision)->GetPropertyOrDefault(
              drake::geometry::internal::kHydroGroup,
              drake::geometry::internal::kComplianceType,
              drake::geometry::internal::HydroelasticType::kUndefined) ==
          drake::geometry::internal::HydroelasticType::kCompliant)) {
        all_geometries.add(collision);
      }
    }
  }
 ```
Thanks to @ggould-tri for asking the question in slack.

DamrongGuoy, 2022-10-12:
Per f2f discussion with @hongkai-dai, different users might have different preferences for what to print, so users should be the ones who print the information. We agreed that we will close this issue.

closed 2022-10-12 by DamrongGuoy

jwnimmer-tri, 2022-10-12:
\CC @trowell-tri FYI

As part of visualizing candidate collision pairs[1], one facet of the information will be which contact mode is in effect for that pair.  Or at least, which contact _properties_ have been annotated on that geometry.

[1] To be clear, there no issue filed about collision-pair visualization yet, but it might be future work at some point -- especially as the planning code moves into Drake.

DamrongGuoy, 2022-10-12:
Part I `Currently in master` of this [G-Doc](https://docs.google.com/document/d/1Q_Mv-LI40fvZQ6h7znUmZlcciM2CQPke2_gpzgaDv9g/edit#heading=h.qjeqk8eob3t0) might be relevant. It's a working document that will update [Hydroelastics Users Guide](https://drake.mit.edu/doxygen_cxx/group__hydroelastic__user__guide.html).

This is the preview of that document.
[image]
[image]