RobotLocomotion/drake #16627: Multiple contact forces between Box and larger Box below it

richardrl, 2022-02-19:
For a small stacking demo, where the top box is initialized with a small orientation offset from the bottom box, there is this strange behavior where there are 2-4 contact forces reported by:

`        ConnectContactResultsToDrakeVisualizer(builder, plant, scene_graph, lcm)
`

when I assume there should only be 1.

https://user-images.githubusercontent.com/5181990/154777307-492a562b-e61f-41e3-9831-2f9db23d8d4f.mp4

richardrl, 2022-02-19:
lcm output:

[image]

jwnimmer-tri, 2022-02-23:
+@sherm1 to assign someone to reply.

sherm1, 2022-03-01:
I can't tell from this issue whether there is something wrong.
@DamrongGuoy can you take a quick look to see if you understand the complaint here? (You looked at a related one in #16634)

DamrongGuoy, 2022-03-14:
@richardrl could you provide the python example to reproduce the problem, please?

DamrongGuoy, 2022-03-22:
I close this issue now since I haven't heard back from @richardrl for a week.

closed 2022-03-22 by DamrongGuoy