RobotLocomotion/drake #16634: Compliant-compliant contact between simple box objects doesn't work

richardrl, 2022-02-20:
https://user-images.githubusercontent.com/5181990/154828280-e2d422f4-86e3-4dae-80a7-2f187d9efbf7.mp4

Here is a minimum reproducible example that only requires pydrake and numpy.
[min_reproducible_hydroelastic_bug.zip](https://github.com/RobotLocomotion/drake/files/8103425/min_reproducible_hydroelastic_bug.zip)

richardrl, 2022-02-20:
@DamrongGuoy Hi Damrong, I saw you were managing this feature. Do you have any insight into what is happening?

DamrongGuoy, 2022-02-20:
Embarrassingly I do not know how to run python.  Could you please tell me steps by steps on how to run your example?  In the past, I only work with C++ examples.  It's time for me to learn python. :smile: 

I quickly looked at `min_reproducible_hydroelastic_bug.py` and do not see settings about hydroelastic modulus (to say how rigid or compliant the boxes are). Would you like it to be like water, plastic, wood, or iron?

A rule of thumb to set hydroelastic modulus is to look up Young's modulus (in Pascals) of the desired material. Then, multiply Young's modulus by the half-thickness (in meters) of the object to get the hydroelastic modulus in Pascals.

For example, polyamide plastic has Young's modulus of 2.5 GPa. Your box is 1 meter wide, so its half thickness is 0.5 meters. Therefore, you might want its hydroelastic modulus to be 0.5 x 2.5e9 Pa = 1.25e9 Pa to simulate two polyamide boxes in contact. If your polyamide box is 2 centimeters wide, its half-thickness is 1 centimeter = 0.01 meter, so its hydroelastic modulus could be 0.01 x 2.5e9 = 2.5e7 Pascals. 

Would [Hydroelastic Contact User Guide](https://drake.mit.edu/doxygen_cxx/group__hydroelastic__user__guide.html) help?

You can teach me python, and I can teach you hydroelastics.  What is this line?
```
from manipulation.scenarios import AddShape, AddFloatingRpyJoint, AddShapeHydro
```

@jwnimmer-tri @rpoyner-tri @sherm1 It might be a good idea to set up a very simple python example (pure python without even SDFormat files).
@richardrl could you tell us a little about your project and yourself, please?

@richardrl we have two draft PRs for hydroelastic examples: #16534 and #15804. I hope to have them in master in a week ot two.

jwnimmer-tri, 2022-02-20:
> You can teach me python, and I can teach you hydroelastics. What is this line?
> 
> from manipulation.scenarios import AddShape, AddFloatingRpyJoint, AddShapeHydro

See https://github.com/RussTedrake/manipulation/blob/master/manipulation/scenarios.py.

richardrl, 2022-02-20:
> Embarrassingly I do not know how to run python. Could you please tell me steps by steps on how to run your example? In the past, I only work with C++ examples. It's time for me to learn python. 😄
> 
> I quickly looked at `min_reproducible_hydroelastic_bug.py` and do not see settings about hydroelastic modulus (to say how rigid or compliant the boxes are). Would you like it to be like water, plastic, wood, or iron?
> 
> A rule of thumb to set hydroelastic modulus is to look up Young's modulus (in Pascals) of the desired material. Then, multiply Young's modulus by the half-thickness (in meters) of the object to get the hydroelastic modulus in Pascals.
> 
> For example, polyamide plastic has Young's modulus of 2.5 GPa. Your box is 1 meter wide, so its half thickness is 0.5 meters. Therefore, you might want its hydroelastic modulus to be 0.5 x 2.5e9 Pa = 1.25e9 Pa to simulate two polyamide boxes in contact. If your polyamide box is 2 centimeters wide, its half-thickness is 1 centimeter = 0.01 meter, so its hydroelastic modulus could be 0.01 x 2.5e9 = 2.5e7 Pascals.
> 
> Would [Hydroelastic Contact User Guide](https://drake.mit.edu/doxygen_cxx/group__hydroelastic__user__guide.html) help?
> 
> You can teach me python, and I can teach you hydroelastics. What is this line?
> 
> ```
> from manipulation.scenarios import AddShape, AddFloatingRpyJoint, AddShapeHydro
> ```
> 
> @jwnimmer-tri @rpoyner-tri @sherm1 It might be a good idea to set up a very simple python example (pure python without even SDFormat files). @richardrl could you tell us a little about your project and yourself, please?
> 
> @richardrl we have two draft PRs for hydroelastic examples: #16534 and #15804. I hope to have them in master in a week or two.

Hi @DamrongGuoy, thanks for your detailed response!

Will spend more time looking into this later today, but my initial investigation of the setting I am using reveals:

AddCompliantHydroelasticProperties(resolution_hint=0.01, hydroelastic_modulus=1, properties=properties)

This is a function I am calling in pydrake.geometry. I’m not sure if the units are the same as you gave, but this is certainly much smaller than those numbers in your post and (may?) explain the behavior.

richardrl, 2022-02-20:
@DamrongGuoy Yep, changing the modulus certainly helped, I guess the low hydroelastic modulus was treating the materials like water. It definitely looks much better than the point contact model in #16627 

It's not perfect when the red box is near the edge of the blue box though. When you drop it, it looks like the red box creates a ball joint at one of the vertices and oscillates with respect to that joint. 

https://user-images.githubusercontent.com/5181990/154864817-972905be-672e-434e-a574-d4b5cb25196e.mp4

By the way, is there an email I can reach you at? Perhaps we can setup a Zoom discussion to setup hydroelastics and Python :)

richardrl, 2022-02-21:
Strangely, something I changed made things way better. If you look at the example script I posted, I set the initial poses with AddShape (from scenarios.py) which calls:
```

plant.RegisterCollisionGeometry(
                body, H,
                pydrake.geometry.Box(shape.width() - 0.001,
                                     shape.depth() - 0.001,
                                     shape.height() - 0.001), name,
                properties)
```

However, I noticed if I initialized the cube in identity orientation and at the origin position, _then_ set the poses, the hydroelastic simulation works much much better.

DamrongGuoy, 2022-02-21:
@richardrl Thank you for the update. I'm glad it's getting better.

I wonder whether `lcm-logger` would be helpful here. Use `lcm-logger` to record LCM messages. Then, use `lcm-logplayer-gui` to go one step at a time, so you can see what it is doing at each time step.
```
damrongguoy@puget-183882-03:~/GitHub/DamrongGuoy/drake (hydro_ball_plate_example)$ bazel build //lcm/...
damrongguoy@puget-183882-03:~/GitHub/DamrongGuoy/drake (hydro_ball_plate_example)$ bazel-bin/external/lcm/lcm-logger
Opening log file "lcmlog-2022-02-21.00"
^CLogger exiting
damrongguoy@puget-183882-03:~/GitHub/DamrongGuoy/drake (hydro_ball_plate_example)*$ bazel-bin/external/lcm/lcm-logplayer-gui lcmlog-2022-02-21.00
```
[image]

If you can set both blocks to be more transparent, we can see the contact surface and the force better.
You might want this visualization settings:

    - Maximum pressure (on the contact surface) 5e6 (or other numbers)
    - Vector scaling mode = Scaled
    - Global scale of all vectors = 0.05 (or other numbers)
[image]

DamrongGuoy, 2022-02-22:
@richardrl Could you please close this issue?  Or do you still have questions?

DamrongGuoy, 2022-02-23:
I close this issue now because I do not hear from @richardrl for two days.  Please feel free to re-open this issue again with more questions.

closed 2022-02-23 by DamrongGuoy