RobotLocomotion/drake #16636: Algebraic loop causes confusing "Recursion limit error"

RussTedrake, 2022-02-20:
Posted on Stack Overflow:
https://stackoverflow.com/questions/70910589/recursion-error-while-evaluating-contact-information
with a small notebook reproducing the error.

The diagram was constructed with MBP contact_results looping directly back to the MBP actuation_input_port as seen in this diagram:
[image]
Something is defeating our attempts to discover the algebraic loop.  One of the systems in the loop is a python leaf system.  

As a result, calling `diagram.Publish(context)` resulted in the confusing error `RecursionError: maximum recursion depth exceeded while calling a Python object`.

The desired outcome would be to provide the "algebraic loop" error message at the time of constructing the diagram.

jwnimmer-tri, 2022-02-20:
Possibly related to #12786?  If so, then there is a fail-fast diagnostic already in place, but possibly not when using point contact?

https://github.com/RobotLocomotion/drake/blob/6e0c78de480f9f6fec4d3ad0448d77f20259cf08/multibody/plant/compliant_contact_manager.cc#L443-L455

xuchenhan-tri, 2022-05-03:
Hmm, I thought the non-contact force calculation in MbP got the same treatment as that in the contact manager. Apparently not...
I'll take a look at this when I get some time.

xuchenhan-tri, 2022-05-12:
Like @jwnimmer-tri said above, the reason that this algebraic loop went undetected is due to #12786 where the non-contact forces' calculation can't properly depend on input ports. But the reason that it's not caught by the existing fail-fast diagnostic is not because of point contact but due to the fact that it doesn't go through the compliant contact manager. I believe routing all non-contact force calculations through the diagnostic should resolve this.

closed 2022-05-16 by xuchenhan-tri