RobotLocomotion/drake #16643: MultibodyPlant Tutorial

xuchenhan-tri, 2022-02-21:
As discussed in #12912, we currently don't have tutorial covering how MbP works. Russ and I started a sketch tutorial in [deepnote](https://deepnote.com/project/Authoring-a-Multibody-Simulation-jnoKyVLkS5CYUgHG3ASsBA/%2Fauthor_multibody_sim.ipynb) that touches on the following aspects mentioned in #12912:

- Authoring and parsing an SDF file.
- Reading and writing a pose.
- Running a simulation with a MbP.
- Visualization.

The tutorial still needs some polish to be considered finished
- There are a couple of TODO's to be filled in. 
- The "Next steps" section needs more work.
- Collision vs. visual geometry can be discussed in more details.
- Link to Doxygen documentation.
- ...

@RussTedrake @jwnimmer-tri

jwnimmer-tri, 2022-04-13:
The probable best workflow here is to finish editing and collaborating on the tutorial using the online Colab notebook, and then once it's in almost-final shape, to copy it into a Drake PR for review and submission into Drake's `drake/tutorials/...` folder, at which point we'd also add it to the `index.ipynb` there.

RussTedrake, 2022-04-16:
fwiw -- i've just updated the deepnote tutorial to use Drake's JointSliders instead of the manipulation joint sliders, and removed the dependency on the manipulation repo.

jwnimmer-tri, 2022-04-20:
@zachfang as this tutorial gets closer to completion, please tag @aykut-tri to help review it for clarity and a fresh set of eyes.  Others of us reviewing it (myself, Xuchen, etc.) are already deeply familiar with MbP; having a mixed-experience audience for review will be super valuable.

jwnimmer-tri, 2022-04-28:
> The probable best workflow here is to finish editing and collaborating on the tutorial using the online Colab notebook, and then once it's in almost-final shape, to copy it into a Drake PR ...

I didn't say this as clearly as I could -- my suggestion is to do the entire Feature Review phase using Deepnote, and only copy it into a Drake PR once that's finished, and we're on to Platform Review.

jwnimmer-tri, 2022-05-06:
I am cleaning up the issues database now.  I'm going to close out #12912 as a duplicate of this issue.  Thus, I'll consolidate some of its suggestions into this issue.

We don't need to act on these immediately.  Once the first pass at this tutorial lands on master, we should revisit all of the ideas above and below, and come up with the goals for a second pass.

From https://github.com/RobotLocomotion/drake/issues/12912#issue-585270143:

- Intro (kinematics, dynamics, topology, etc). Mostly, this should point to Doxygen / Underactuated textbook, and not repeat too much?
- Parsing models. Should also cover parsing multiple models, and model instances.
- Querying joints, bodies, etc. Should cover model instances (e.g. errors you get when you have multiple of the same name).
- Minimal coordinates (positions, velocities). Read/write. Talk about floating-body base coordinates?
- Maximal coordinates (pose). Read/write. CalcRelativeTransform, etc.
- Visualization (relates #12645, though perhaps we could do 2D projections?), first w/o sim.
- Dynamics.
- Basic geometry, collision. (We may want to save complex bits for a separate tutorial?)
- Simulation. Sensors, force
  - Also cover purely kinematic simulations? (e.g. driving kinematics by input ports to position / velocity).

From https://github.com/RobotLocomotion/drake/issues/12912#issuecomment-602067832:

- Creating your multibody system from urdf/sdf.  Simple geometry inspector interface with jupyter joint sliders and an embedded meshcat visualizer
- Core design: MBP + SceneGraph
- Simulation.  
  - Visualizers.  Playback.
  - Simulator as a ROS node
- Kinematics and dynamics.  Show symbolic.
- (Inverse) kinematics.  Mention global IK
- Call out to the existence many other advanced algorithms (for motion planning, etc)

From https://github.com/RobotLocomotion/drake/issues/12912#issuecomment-602138523:

- I am modifying IK API as mentioned in https://github.com/RobotLocomotion/drake/issues/12849#issuecomment-598848762, I will add the IK tutorial after the API is stabilized.

jwnimmer-tri, 2022-05-09:
Note that I've filed #17132 for a related, collision-specific tutorial.

RussTedrake, 2022-06-12:
This seems like the right issue to add one more idea/request:

It would be nice if the authoring multibody plant tutorial could also show people how to author and visualize kinematic frames.  I wrote the following utilities, which I find super valuable for this workflow:
https://github.com/RussTedrake/manipulation/blob/346038d7fb3b18d439a88be6ed731c6bf19b43de/manipulation/scenarios.py#L367-L421

RussTedrake, 2022-06-12:
Also, from slack:

[russtedrake](https://tri-internal.slack.com/archives/C02KXJ6LM54/p1654939112447029?thread_ts=1651202317.475869&cid=C02KXJ6LM54)
in this tutorial, there are a few examples of having the sdf string directly in the jupyter cell... a workflow I like very much.  But then you write it to disk and then load it with AddModelFromFile.  Why not just AddModelFromString??  I don't mind changing it, but thought I would avoid Chesterton's fence.

[jeremy.nimmer](https://tri-internal.slack.com/archives/C02KXJ6LM54/p1654956738627039?thread_ts=1651202317.475869&cid=C02KXJ6LM54)
Discussed during review at https://github.com/RobotLocomotion/drake/pull/17148#pullrequestreview-971126542.  I'd support moving model_inspector into pydrake proper and accepting either filename or string.

jwnimmer-tri, 2022-11-09:
> I'd support moving model_inspector into pydrake proper and accepting either filename or string.

For the record, that's what #18249 is doing.

jwnimmer-tri, 2026-02-06:
We have a lot of tutorials now.  If there are still missing pieces, let's open more focused new issues for those.