RobotLocomotion/drake #16676: Drake's realsense2_description is unused

jwnimmer-tri, 2022-02-25:
Per https://github.com/RobotLocomotion/drake/pull/14080#pullrequestreview-490046631:

> I plan to add it to the manipulation station, and also use it in many small python examples that involve the cameras and objects (but not the entire station).

As best I can tell, nowhere in Drake, underactuated, nor manipulation has started using the these realsense2_description files yet.

@RussTedrake is your plan still to add a realsense2 to the manipulation_station?

I ask because its `package.xml` is messed up (and we never noticed), so we're either going to need to fix the xml file, or remove the realsense2.

jwnimmer-tri, 2022-02-26:
Hopefully not too difficult.  We'd need to:

(1) Remove `drake/manipulation/models/realsense2_description/package.xml` from our runfiles and installed files tree:

https://github.com/RobotLocomotion/drake/blob/246778b8d4124f689eb7b5eb2beaa8a130d38bba/tools/workspace/intel_realsense_ros/files.bzl#L10

(2) Replace `package://realsense2_description` to say `package://drake/manipulation/models/realsense2_description` in the urdf instead, possibly at sed-time:

https://github.com/RobotLocomotion/drake/blob/246778b8d4124f689eb7b5eb2beaa8a130d38bba/tools/workspace/intel_realsense_ros/repository.bzl#L19-L22

jwnimmer-tri, 2022-03-03:
> I've been using a small realsense-shaped box so far ...

I figured I would just go fix this since it should be easy, but I got stuck trying to find the box that should be replaced with this camera mesh.  Do you have any pointers to where I should change?

Or is the request that `ManipulationStation::Finalize` should add illustration geometry for the cameras in `camera_information_`?  It seems to only add the sensor modules, not any geometry.

RussTedrake, 2022-03-05:
Thanks.  I have a placeholder https://github.com/RussTedrake/manipulation/blob/master/manipulation/models/camera_box.sdf that gets loaded from a few places.  My aim is to replace that we the realsense geometry.

jwnimmer-tri, 2022-03-05:
Is there anywhere inside of Drake itself where you'd like us to use this model?

Drake is not a model repository for other projects to use; any models in Drake's build system [must be referred to by some example or demo](https://drake.mit.edu/model_version_control.html).

If this model won't be used in Drake, then I'd rather focus our time on issues like #9498 or #15024 instead, so that you can point your manipulation repository at https://github.com/IntelRealSense/realsense-ros directly, without laundering the files through Drake.

RussTedrake, 2022-03-05:
Right now the manipulation station has `MakeD415CameraModel` but does not actually add the camera geometry.  It should.

The reason I didn't recommend that immediately, is because it's not as drop in of a replacement and because of #14992.  I consider the ManipulationStation the manipulation repo the current best version.

closed 2022-03-08 by sammy-tri