RobotLocomotion/drake #16738: SAP Solver PR train

amcastro-tri, 2022-03-09:
This issue provides a sketch of the PR train needed to complete the [SAP solve project](https://github.com/RobotLocomotion/drake/projects/10). Refer to the project's description for more information.

Find [SapSolver project here](https://github.com/RobotLocomotion/drake/projects/10). Full working prototype in https://github.com/RobotLocomotion/drake/pull/16543.

- [ ] [SapContactProblem](https://github.com/RobotLocomotion/drake/pull/16548) 
  - Without SapContraint
  - Without ContactProblemGraph
  - Define cliques
- [ ] [SapConstraint](https://github.com/RobotLocomotion/drake/pull/16553)
  - Reference SapContactProblem's docs on cliques
- [ ] [ContactProblemGraph](https://github.com/RobotLocomotion/drake/pull/16570)
- [ ] [SapContactProblem::AddConstraint()](https://github.com/RobotLocomotion/drake/pull/16618)
- [ ] [SapConstraintBundle](https://github.com/RobotLocomotion/drake/pull/16697)
- [ ] [PartialPermutation](https://github.com/RobotLocomotion/drake/pull/16596)
- [ ] [Contact graph provides participating cliques](https://github.com/RobotLocomotion/drake/pull/16694)
- [ ] [FrictionConeConstraint](https://github.com/RobotLocomotion/drake/pull/16715)
  - Use MP to get a numerical solution to the projection.
- [ ] [Update MultibodyTreeTopology to provide tree information needed by the manager.](https://github.com/RobotLocomotion/drake/pull/16725)
  - num_tree_velocities_: Needed to compute per-tree A and J blocks.
  - body_to_tree_index_: Needed to determine clique given a body in contact.
  - tree_velocities_start_: needed to extract per tree Jacobian blocks.
  - dof_to_tree_index_: Thus far only used when adding distance constraints. I'll just add all these in a single PR.
- [ ] [SapModel](https://github.com/RobotLocomotion/drake/pull/16713)
- [ ] Complete SapModel features + unit tests:
  - Regularizer cost
  - Gradient of the cost.
  - Hessian of the cost
  - United with 2D boxes with contact and AutoDiff for gradients.
- [ ] CompliantContactManager makes SapContactProblem
  - Eval contact Jacobian as needed.
  - Eval? SapContactProblem?, a function of state after all.
- [ ] SapContactSolver is no longer a ContactSolver
- [ ] SapSolverResults
  - Introduce results compatible with a given SapContactProblem
  - Extract ContactResults utility for contact reporting.
- [ ] SapSolver makes use of SapModel
- [ ] (stretch) SapSolver unit tests
  - Push box
  - Falling cylinder
  - Painleve rod
  - Coupler and distance constraints.
- [ ] CouplerConstraint
- [ ] Incorporate SapSolver into CompliantContactManager
  - Testing strategy?
- [ ] Manager API to add CouplerConstraint
- [ ] MultibodyPlant option to use SapSolver
  - MBP API.
  - MultibodyPlantConfig
    - solver option
    - regularization params option?
  - Add gflag to opt-in for SAP in some of our examples:
    - Falling cylinder
    - Acrobot
    - Atlas
    - ManipulationStation
- [ ] Parsing of dissipation time scale for SAP dissipation model
- [ ] PD controller constraint
  - Add it to compliant contact manager in the same PR?

sherm1, 2022-03-09:
Should we add "support joint locking" to this list?

DamrongGuoy, 2022-03-09:
It is so great to see the plan. Thank you!

edrumwri, 2022-09-09:
Could I make two requests?

1. Could you bump the priority for joint locking?
2. Would you cause SAP solver to fail fast if joint locking is enabled? We spent three hours of engineering time tracking down why our locked bodies were no longer stationary.

chen-tianjian, 2023-01-07:
May I ask what the status is for joint locking and free body locking with SAP?

jwnimmer-tri, 2024-11-02:
@amcastro-tri This is finished, I think?

closed 2024-11-02 by jwnimmer-tri