RobotLocomotion/drake #16750: RenderRpc: documentation of the HTTP protocol

zachfang, 2022-03-09:
Finalize the API documentation for the HTTP protocol between client/server communication.

svenevs, 2022-03-17:
Note: #16748 and #16749 are related here, the documentation needs to be converted to doxygen and link to where those files are going to live.  Current docs are here: https://github.com/RobotLocomotion/drake/blob/85135d0be06ef4bb45265fc0138b347cedd1191d/geometry/render/dev/render_server/README.md

svenevs, 2022-03-24:
FYI when things move out this documentation should link to the final destination

https://github.com/RobotLocomotion/drake/blob/06dd087b402438853f3227aaa8e73889ee7d09ac/geometry/render/render_doxygen.h#L19-L24

zachfang, 2022-11-10:
Close by #18126. There is a potential edit on how we render label images, but the documentation reflects the current code and is complete.

closed 2022-11-10 by zachfang