RobotLocomotion/drake #16784: SDF parser: support first-class feedback channel

rpoyner-tri, 2022-03-14:
Split from https://github.com/RobotLocomotion/drake/issues/16229. Proposal is to plumb the full SDF parser via common/diagnostic_policy, so that handling of errors and warnings can be configured.

Victory condition: all instances of throw, drake::log->warn(, and log_once reachable from AddModelsFromSdf() are replaced by diagnostic policy usage. There may be specific exceptions for difficult cases.

jwnimmer-tri, 2022-04-26:
Given the implementation tactic uncertainty in #16785, I'm going to change the victory condition here.  Re-spelling the plumbing to use "diagnostic.Error" vs "throw" is not terribly useful.  We are not certain that the existing call paths are were we're going to detect new kinds of errors.  If we find it necessary, we can add new plumbing as-needed.

The new victory conditions are:
- [ ] All warnings are performed via `diagnostic.Warn`, instead of bespoke mechanisms.  That allows promotion to errors, which is an important part of the use case.
- [ ] Change any errors (or exceptions) that should be warnings instead to use `diagnostic.Warn`.

EricCousineau-TRI, 2022-04-27:
Per Slack DM, in Anzu PR 8769, r1, I could see super useful filenames + lineno. for URDF files, but not for SDFormat files.
Any chance that could be part of this work? (and might you want me to work w/ OSRC on this?)

Most notably, this error message appeared w/o direct trace back to files (which were SDFormat):
https://github.com/RobotLocomotion/drake/blob/da38019a96d7015628be1f620c0fb00aa97307c0/multibody/parsing/detail_common.cc#L79-L83

jwnimmer-tri, 2022-04-27:
Our original intention for the scope of this issue was to have the parsing code use a `diagnostic` object at all to report messages, instead of bespoke `throw` or `drake::log()` or etc. -- as kind of a baseline of infrastructure.  (And I've even down-scoped it since that point.)

We didn't intend to try to update all of the `diagnostic` calls to also add in the filename or line number.  Doing so would definitely be useful, but probably worth its own separate issue.  If OSRC wanted to jump in and work on that, I think it would be very helpful.

EricCousineau-TRI, 2022-04-27:
Sounds good! --> #17053

PR #17054 (2022-04-29): [parsing] Slightly improve joint parsing diagnostics
Mostly, just clarify our plans for future work, and be sure to notice when SDFormat adds new joint types (via C++ switch-enum warnings).

To provide more context for the TODOs -- as discussed in slack, to the extent that we can best-effort deal with unimplemented features (while warning about it to the user), that is probably the best UX.  It will allow the user to actually create a MbP and play with it, resolving the warnings later in due course.

Closes #16784.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/dra [... cut]
closed 2022-04-29 by xuchenhan-tri