RobotLocomotion/drake #16796: Investigate Fission builds on Mac

BetsyMcPhail, 2022-03-15:
Follow up to #16500.

On Mac, the goal is to enable fission to reduce disk space on debug CI builds. The first task here is to determine if this option will be helpful.

Also see #16788

jwnimmer-tri, 2022-03-21:
I just realized that Debug builds on macOS run only Weekly in the CI schedule:
https://drake-jenkins.csail.mit.edu/view/Mac%20Monterey/job/mac-monterey-clang-bazel-weekly-debug/

Because they aren't running that often (I'd assumed they were nightly), if it ends up being too much work to experiment with Fission for them, we can probably just WONTFIX here and give up.

mwoehlke-kitware, 2022-03-23:
As a checkpoint, I ran `bazel test --config=apple_debug //common/...`. The resulting `bazel-out/` was 4.5G. (No change with or without `--fission`.)

mwoehlke-kitware, 2022-04-07:
I'm not entirely sure how much I trust CI disk usage reports. Nevertheless, I used #16907 to run a build. Here are the results:

### Without `--fission`
```
[8:39:21 PM]   Filesystem     Size   Used  Avail Capacity iused    ifree %iused  Mounted on
[8:39:21 PM]  /dev/disk1s1  160Gi  133Gi  1.4Gi    99% 1396627 14556240    9%   /System/Volumes/Data
```

### With `--fission`
```
[4:18:43 PM]   Filesystem     Size   Used  Avail Capacity iused     ifree %iused  Mounted on
[4:18:43 PM]  /dev/disk1s1  200Gi  138Gi   45Gi    76% 1399652 472804240    0%   /System/Volumes/Data
```

Note that "without `--fission`" was an old build, so this isn't exactly an apples-to-apples comparison. Nevertheless, as I understand it, we would expect to see a non-trivial decrease in disk usage if this change is desirable.

BetsyMcPhail, 2022-04-11:
I found a newer build (https://drake-jenkins.csail.mit.edu/view/Mac%20Monterey/job/mac-monterey-clang-bazel-weekly-debug/14) reports:

```
[6:36:56 PM]    *** Disk usage for /tmp:
[6:36:56 PM]   Filesystem     Size   Used  Avail Capacity iused     ifree %iused  Mounted on
[6:36:56 PM]  /dev/disk1s1  200Gi  138Gi   45Gi    76% 1350485 467591840    0%   /System/Volumes/Data
```

jwnimmer-tri, 2022-04-11:
OK, seems like no real gain here.  Let's give up for now.

closed 2022-04-11 by jwnimmer-tri