RobotLocomotion/drake #16811: Feature request: reuse optimization workspace and problem setup

mposa, 2022-03-16:
In some optimization settings, MPC for example, similar programs are solved sequentially. Particularly for control, throughput can have a substantial effect on closed-loop outcomes. In Drake's currently implementations of solvers for `MathematicalProgram`, upon calling `Solve`, the code currently creates the solver-specific workspace and problem definition, solves the program, and then destroys the workspace. Depending on the size of the program, there is substantial overhead in workspace creation that could easily be eliminated by reusing the workspace and problem setup.

In our code, @yangwill implemented something like this for OSQP https://github.com/DAIRLab/dairlib/blob/master/solvers/fast_osqp_solver.h.

It would be great if this idea were picked up on a broader scale, for more solvers, as part of Drake proper.

jwnimmer-tri, 2022-03-16:
A while back, I started a small prototype of this at https://github.com/jwnimmer-tri/drake/commits/osqp-heap.

We cannot store the workspace in the Solver object itself, because those objects must remain thread-safe.  Probably the design we want to aim for is to store the reusable workspace inside the `MathematicalProgramResult`, as is done on that branch.

jwnimmer-tri, 2022-03-23:
One interesting design question here, though -- should we expect that the solvers _notice_ that the problem shape is the same when restarting, or are we thinking about some API where the user tells / promises to us something about the program?

EricCousineau-TRI, 2023-05-29:
Tinkered with warm-starting (but not workspace preservation) in #19498. This is to save on solver iterations, but not memory allocations.
In that prototype, I did want to preserve the dual solution without passing in additional arguments.

In this case, I made a "general" `drake_warm_start` flag.
If set to true, `SolverBase::Solve` will not reset the solve data via `*result = {}`.
It's not perfect, 

For things like workspace preservation, perhaps the solver can swap any post-solve data out (like `workspace` in the prototype), reset the details, then place the post-solve data back in.
(That may still incur some malloc, e.g. in `Eigen::VectorXd y`).

FranekStark, 2023-06-20:
Having just worked a lot with MPC problems in drake and other solving frameworks, I would also like to give my thoughts on this.
I think it is definitely necessary to implement this feature to make drake efficient for mpc. I have run some tests and depending on the solver, the pure creation of the problem (i.e. the matrices etc.) sometimes takes longer than the solving. Therefore, in my case, it was sometimes more efficient to reformulate the problem into large matrices beforehand and then add only one constraint. In the next control step, I then efficiently changed this large matrix at the appropriate points and passed it to drake. 

To address @jwnimmer-tri  point, the following workflow would make sense for me as a user:
The mathematical problem is created as before. So the user adds constraints etc. When solve is called, the problem is transformed into the interfaceable matrices for the respective solver and then solved. As others have already suggested, these matrices would have to be kept somewhere. The problem is therefore in the “transformed state”. Depending on the solver, in the solver itself or in drake's solver interface (some solvers are very efficient in this respect). Now there are methods that change the mathematical problem. Here we would have to distinguish between methods that change the size of the problem or only change values. Value-changing methods can be permitted in this "transformed" state, and depending on the solver, the matrix or even only a single coefficient is changed in the solver interface or directly in the solver. The mapping would still have to be maintained somehow, but I think it can be done well.