RobotLocomotion/drake #16814: RenderRpc: promote test_utilities to Drake master

svenevs, 2022-03-17:
Go through the Drake platform review to merge `geometry/render/dev/test_utilities` for dummy PNG / TIFF images code to master.

Relates: #16802.

jwnimmer-tri, 2022-05-28:
We've talked about reworking the unit tests to perhaps stop using these utilities.

However, if we _do_ keep these functions around then we will need to rewrite them, as found by a dependency audit I'm working on for #17231.  Drake should _not_ be calling libpng nor libtiff directly from all over our codebase. If we need to do image file I/O, we can use vtkImageIO for that purpose.  It's probably just as easy to commit a sample tiff and png image into git, though, instead of generating them on the fly.

svenevs, 2022-05-28:
FYI see `y - 1` comment https://github.com/RobotLocomotion/drake/pull/16751#pullrequestreview-904996548 (and a bit lower `something that is not regular / all constant values is a good test case`).  You could easily generate using current code and then track the file, just don't do all uniform.

However, the tests were implemented the way they were (using the underlying libpng and libtiff) because the whole point is to test whether or not vtk is loading the image as expected (e.g., drake and vtk have [different notions of (0, 0) image corner](https://vtk.org/doc/nightly/html/classvtkImageReader2.html#a3a959f0dbfb4ff4b0227317941d7ea19), IIRC tiff and png behave differently with whether or not lower left gets set (I think tiff has metadata to state the corner?  I forget).  Using vtk to test whether vtk loaded the image correctly was the original implementation, and why I never found the `y - 1` offset.  That's the underlying "dilemma" here -- transforming vtk image loaded memory into drake buffers, render engine vtk uses vtkImageExport but we can't do that here (e.g., client sends RGB but destination drake buffer is RGBA, similar issues with tiff IIRC).

The dependency audit doesn't mention libtiff or libpng (yet) and right now they come from package managers, but it sounds like you want to pave the path for them to be private.  It's fixed now and very unlikely to be a problem in the future.  Instead of using VTK or tiff or png to compare, you could probably just transform these to return the final desired buffer in memory as a drake image, and track the generated images in git (probably with a comment linking to a commit for how to re-generate them if needed).  That gets you away from using libpng / libtiff directly, but still lets you effectively compare the expected memory layout (I think).

jwnimmer-tri, 2022-06-02:
Zach and I chatted f2f on this.  The plan is to commit sample images to git (smaller HxW ones than the test currently uses) and then we can hard-code the unit tests for look for those specific files' pixels.  In other words, we won't generate test images on the fly anymore.

PR #17449 (2022-07-19): [geometry/render_gltf_client] add RenderClient class
This PR moves the` RenderClient` class and the image testing infrastructure from the dev folder to `//geometry/render_gltf_client/`. RenderClient is the central part of a client that sends out the POST request, decodes the server response, and loads the image for a Drake simulation.

It also contains some reworking of the URL logic discussed in #16950.

A high-level project overview can be found [here](https://drake.mit.edu/doxygen_cxx/group__render__engine__gltf__client__server__api.html).

Close #16814 
Close #16746

---
This change is [[i [... cut]
closed 2022-07-19 by jwnimmer-tri