RobotLocomotion/drake #16832: GitHub Issue Template Outdated

svenevs, 2022-03-22:
Visit in browser for a warning from GitHub: https://github.com/RobotLocomotion/drake/blob/master/.github/issue_template.md

They link to [the new format](https://docs.github.com/en/communities/using-templates-to-encourage-useful-issues-and-pull-requests/about-issue-and-pull-request-templates).  Some reference examples:

- https://docs.github.com/en/communities/using-templates-to-encourage-useful-issues-and-pull-requests/syntax-for-issue-forms
- https://github.com/pytorch/pytorch/tree/master/.github/ISSUE_TEMPLATE You can create a new issue there (to then just cancel out of) to see how the forms work.

Low priority, unclear when `.github/issue_template.md` will no longer be supported.

CC @EricCousineau-TRI as the original author of the issue template and possible vested interest, but no direct assignment :wink:

EricCousineau-TRI, 2022-03-25:
No direct interest in completing :sweat_smile: 
Per [Platform Reviewer Checklist](https://drake.mit.edu/platform_reviewer_checklist.html), assigning to @BetsyMcPhail for disposition.

EricCousineau-TRI, 2022-06-09:
Closed via #17372

closed 2022-06-09 by EricCousineau-TRI