RobotLocomotion/drake #16841: DrakeVisualizer: Cannot Toggle Hydroelastic Contact Visualizer

rcory, 2022-03-22:
When toggling off hydroelastic contact visualization in Drake Visualizer through the tools menu (`Tools->Hydroelastic Contact Visualizer`), there is no way to toggle it back on. That is, once the hydro visualization is toggled off, it remains off until Drake Visualizer is re-started again.

DamrongGuoy, 2022-03-22:
I can reproduce the problem as shown in these pictures.

[image]
[image]
[image]
[image]

DamrongGuoy, 2022-03-22:
I see this error message:
```
Traceback (most recent call last):
  File "/home/damrongguoy/.cache/bazel/_bazel_damrongguoy/aa787bc96edbb97f62958306ba5638ee/external/drake_visualizer/lib/python3.8/site-packages/director/lcmUtils.py", line 119, in handleMessage
    callback(msg)
  File "/home/damrongguoy/.cache/bazel/_bazel_damrongguoy/aa787bc96edbb97f62958306ba5638ee/execroot/drake/bazel-out/k8-opt/bin/tools/drake_visualizer.runfiles/drake/tools/workspace/drake_visualizer/_drake_visualizer_builtin_scripts/show_hydroelastic_contact.py", line 1160, in clear_on_load
    self.visual_model.clear()
  File "/home/damrongguoy/.cache/bazel/_bazel_damrongguoy/aa787bc96edbb97f62958306ba5638ee/execroot/drake/bazel-out/k8-opt/bin/tools/drake_visualizer.runfiles/drake/tools/workspace/drake_visualizer/_drake_visualizer_builtin_scripts/show_hydroelastic_contact.py", line 769, in clear
    body_contact.clear()
  File "/home/damrongguoy/.cache/bazel/_bazel_damrongguoy/aa787bc96edbb97f62958306ba5638ee/execroot/drake/bazel-out/k8-opt/bin/tools/drake_visualizer.runfiles/drake/tools/workspace/drake_visualizer/_drake_visualizer_builtin_scripts/show_hydroelastic_contact.py", line 696, in clear
    om.removeFromObjectModel(self._folder)
  File "/home/damrongguoy/.cache/bazel/_bazel_damrongguoy/aa787bc96edbb97f62958306ba5638ee/external/drake_visualizer/lib/python3.8/site-packages/director/objectmodel.py", line 593, in removeFromObjectModel
    _t.removeFromObjectModel(obj)
  File "/home/damrongguoy/.cache/bazel/_bazel_damrongguoy/aa787bc96edbb97f62958306ba5638ee/external/drake_visualizer/lib/python3.8/site-packages/director/objectmodel.py", line 385, in removeFromObjectModel
    self._removeItemFromObjectModel(self._getItemForObject(obj))
  File "/home/damrongguoy/.cache/bazel/_bazel_damrongguoy/aa787bc96edbb97f62958306ba5638ee/external/drake_visualizer/lib/python3.8/site-packages/director/objectmodel.py", line 257, in _getItemForObject
    return self._objectToItem[obj]
KeyError: <director.objectmodel.ContainerItem object at 0x7fd3c6dc8790>

```

DamrongGuoy, 2022-03-23:
CC: @rpoyner-tri do you see something obviously wrong?

The ability to toggle hydroelastic visualizer on and off would be very handy.  I hope we can fix it easily.

rpoyner-tri, 2022-03-23:
It's probably worth a look, but don't forget that drake-visualizer is reaching end-of-life soon. So it's only worth effort until the replacement meshcat solution becomes available.

DamrongGuoy, 2022-03-23:
I agree.  We should fix it only if it's easy.

I ask for high priority for the Onsite demo next week.

rpoyner-tri, 2022-03-23:
Had a f2f with @joemasterjohn . I think he will be able to offer some useful advice before too long. I will bow out for  now. :)

joemasterjohn, 2022-03-23:
One question I had before digging deep into a solution for this is: do you just want to hide all of the hydroelastic contact visualizations, or is there a need to disable the plugin itself? Because if all you need to do is hide the visuals, you can just toggle the eye icon for "Hydroelastic Contact" in the Scene Browser:

[image]

DamrongGuoy, 2022-03-23:
That's a very good question, @joemasterjohn.  I let @rcory answer that.

I add that the `eye` works for one snapshot of LCM message.  In the next publish step, the next hydroelastic contact result will show up again.  In other words, the `eye` opens itself for the next message.

joemasterjohn, 2022-03-23:
Yeah the fact that the `eye` button re-enables for each new LCM message is frustrating.

Regardless, I have a fix for the original problem in #16852. @rcory @DamrongGuoy could you please verify that this fixes the problem for you? Thanks.

rcory, 2022-03-23:
Verified that the associated PR fixes the issue for me. Thanks!

PR #16852 (2022-03-24): [geometry:drake_visualizer] Fix for disabling/enabling hydro contact plugin
Fixes #16841

Fixes a bug that was improperly clearing the data model for the hydroelastic plugin, resulting in disabling/re-enabling the plugin not showing hydro contact surfaces.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/16852)
closed 2022-03-24 by joemasterjohn