RobotLocomotion/drake #16842: Dropping Ubuntu 18.04 (Bionic) CI

jwnimmer-tri, 2022-03-22:
The first task in #13391 is to turn off CI jobs for Bionic.

- [ ] At https://drake-jenkins.csail.mit.edu/
- [ ] At https://github.com/RobotLocomotion/drake-external-examples/tree/main/.github

Feel free to start landing this work on 2022-04-01 even without TRI reviews (we'll be out).  The only gating item is that we should be sure that Drake v1.1.0 gets released before we shut down Bionic.

\CC @rpoyner-tri FYI as release manager for v1.1.0.

BetsyMcPhail, 2022-04-07:
Note, the current pre-merge builds are:

`linux-bionic-clang-bazel-experimental-everything-release`
`linux-bionic-clang-bazel-experimental-release`
`linux-bionic-clang-cmake-experimental-release`
`linux-bionic-gcc-bazel-experimental-debug`
`linux-bionic-gcc-bazel-experimental-everything-release`
`linux-bionic-gcc-bazel-experimental-release`            
`linux-bionic-gcc-cmake-experimental-release`
`linux-focal-clang-bazel-experimental-leak-sanitizer`
`linux-focal-gcc-bazel-experimental-release`

`linux-bionic-gcc-bazel-experimental-release` will be a duplicate when switched to focal, so there will be 8 pre-merge builds after the switch.

BetsyMcPhail, 2022-04-08:
I confirmed that the Bionic CI jobs, in general and production, are a subset of the Focal jobs.

BetsyMcPhail, 2022-04-11:
@jwnimmer-tri last time we changed pre-merge jobs it required some intervention from you on Github (https://drakedevelopers.slack.com/archives/C270MN28G/p1644614660867079). I have the Jenkins changes ready to go, let me know when would be a good time for you to do the switch.

jwnimmer-tri, 2022-04-11:
GitHub's project settings have the new builds as required on pull requests now:

[image]

BetsyMcPhail, 2022-04-11:
Bionic jobs have been removed from Jenkins and CI no longer supports Bionic ([PR #146](https://github.com/RobotLocomotion/drake-ci/pull/146))

closed 2022-04-14 by jwnimmer-tri