RobotLocomotion/drake #16848: Use of SceneGraphInspector requires use of drake::geometry::internal

ggould-tri, 2022-03-23:
Related to #16615 (and possibly fixed by any fix to that issue?)

The public API of `SceneGraphInspector` provides `GetProximityProperties`, which gives a string-based map of properties.  However both the keys and the values in that map are located in `drake::geometry::internal`, forcing downstream users to write code like:
```
using drake::geometry::internal::HydroelasticType;
inspector.GetProximityProperties(collision)->GetPropertyOrDefault(
              drake::geometry::internal::kHydroGroup,
              drake::geometry::internal::kComplianceType,
              HydroelasticType::kUndefined)
```
Every use of the word "internal" in that snippet is a bug, since downstream users should not be concerning themselves with internal (and, by extension, unstable) APIs.
If this material is truly internal, it needs external accessors that present the API in its intended form, whatever that may be.  If it is not truly internal we shouldn't be using an internal namespace for it.

rpoyner-tri, 2022-03-23:
I'll keep it for now. F2f with @ggould-tri indicated this is not an on-fire request, but I agree it should be phased out in some way.
One possible path forward is outlined in #13688.

EricCousineau-TRI, 2023-02-24:
I just ran into this asymmetry as well.

Per @DamrongGuoy's suggestion re: #18241, I would like to (naively) make all of my geometries be compliant to avoid instability. As a first pass, in `pydrake`, I'd like to say, "if this isn't hydro-complaint, make it so", for initial debugging.
However, my hands are tied in Python due to the property being internal.

While #13688 looks great, it does look non-trivial.
Is it possible to hoist these relevant properties to public? And perhaps even the string constants? (assuming they haven't changed w/in the last year?)

As a workaround, I will write some C++ code in Anzu to provide the "is it hydro-compliant" check, bind that, and then I can leverage existing public API.

SeanCurtis-TRI, 2023-02-27:
Some thoughts on the simple fix:

1) Simply making the `kPropertyThing` public is path forward. However, `HydroelasticType::kSoft` should be `HydroelasticType::kCompliant` prior to doing that. The term is no longer consistent with the language we use but was left as is because it was internal.
2) It would be *really* nice to get #13688 in before setting these kind of details loose in public namespaces.

As an alternative workflow using current APIs:
`SceneGraphInspector` has `SceneGraphInspector::maybe_get_hydroelastic_mesh()`. If it has a hydro representation, you'll be able to infer rigid/compliant by the state of the variant.