RobotLocomotion/drake #16865: Upgrade to Bazel 5.1 in Provisioned

jwnimmer-tri, 2022-03-24:
https://github.com/bazelbuild/bazel/releases/tag/5.1.0

Prior issue: #16447.
Prior PR: #16523.

No hurry on this, it's OK for this to wait until the 2022-04-01 externals upgrade / bionic drop time period.  I'll be running it locally in the meantime and will chime in about any local problems I see.

BetsyMcPhail, 2022-03-28:
Note that Homebrew is still at 5.0.0

BetsyMcPhail, 2022-03-28:
- [ ] Upgrade recommended Bazel version to 5.1.0 for Linux - #16878
- [ ] Update Linux provisioned images - Coordinate with update with #16864
- [ ] Upgrade recommended Bazel version to 5.1.0 for Mac - waiting for Homebrew
- [ ] Update Mac provisioned images

mwoehlke-kitware, 2022-03-31:
...wheels? (macOS wheel will follow homebrew / CI provisioning, but the Linux wheels need to be updated.)

FYI, I'm running a local wheel build with 5.1.0 now... probably going to take a few hours since this invalidated all of the caching. (Would there be any value in moving Bazel provisioning to after dependencies but before Python provisioning?)

BetsyMcPhail, 2022-03-31:
@mwoehlke-kitware [tools/wheel/image/provision-base.sh](https://github.com/RobotLocomotion/drake/pull/16878/files#diff-1d9ab263d143c0d3b9bbb66f3f1017a6c45b3100cb893ac386f3775945cfd8fe) was updated as part of the Linux PR. Where else do I need to update? I probably missed it last time as well so we may be 2 versions out of date.

mwoehlke-kitware, 2022-03-31:
Ah, apparently my local checkout is that far out of date 😢 .

PR #16926 (2022-04-11): [doc] Upgrade recommended Bazel version to 5.1 on Mac
Resolves #16865.

The unprovisioned Mac builds switched to 5.1 over the weekend. For example (search for bazel 5.1 in the console ouput):

https://drake-jenkins.csail.mit.edu/view/Mac%20Monterey%20Unprovisioned/job/mac-monterey-unprovisioned-clang-cmake-weekly-release/14/consoleFull
https://drake-jenkins.csail.mit.edu/view/Mac%20Big%20Sur%20Unprovisioned/job/mac-big-sur-unprovisioned-clang-bazel-nightly-release/427/consoleFull

The provisioned Mac images are being updated now.

---
This change is [[image]](https://reviewable.io/reviews/robo [... cut]
closed 2022-04-11 by BetsyMcPhail