RobotLocomotion/drake #16882: Building as a static library

aminya, 2022-03-29:
It seems that Bazel doesn't allow building a library as a static library by default, which is unfortunate as it means the users should either use the shared library or use Bazel as their build system. 

There is an issue for this in Bazel.
https://github.com/bazelbuild/bazel/issues/1920

There is one solution offered in the end. I tried it, but it lacks the libraries that Drakes uses (gflags).
https://github.com/bazelbuild/bazel/issues/1920#issuecomment-1019661738

jwnimmer-tri, 2022-03-29:
This sounds like a feature request on Drake, but I'm not sure exactly what feature you're requesting, or what problem you're trying to solve.  Could you try to explain your desired result?

Would it solve the problem if when doing `cmake && make install` for Drake, that there was an option `-DDRAKE_BUILD_STATIC=ON` (off by default) that would install `libdrake.a` instead of `libdrake.so`?

If that's along the right path, then what should happen with the dependencies that are currently dynamically linked?  Some of them are LGPL (LCM, dReal/IBEX), so we would either need to continue dynamically linking them, or else disable them.

Similarly, when building statically, we'd need to disable pydrake (since Python loadable native modules are basically always dynamic).

jwnimmer-tri, 2022-03-31:
Ok, this makes sense.  The feature request is for our CMake build to obey `BUILD_SHARED_LIBS` to the extent possible.

We'll need to do #16780 first (so that we have a C++ standalone build, with no Python), and then we can return to this issue.

jwnimmer-tri, 2025-07-14:
From f2f: probabe new bazel `config_flags`:
- opt-out of installing pydrake
- switch libdrake from `.so` to `.a`

On the bazel side, it's OK for these to be "internal use only".

tyler-yankee, 2025-08-01:
For reference if this ends up being picked up by someone else - my WIP branch is here: https://github.com/tyler-yankee/drake/tree/static-lib.

closed 2025-09-26 by tyler-yankee