RobotLocomotion/drake #16890: CMake undefined reference build errors

aminya, 2022-03-31:
I am trying to use the library built with CMake, however, my build fails with missing reference errors. I have tried several things, but it seems that `schema` references are missing from libdrake.so. This issue doesn't happen if I use Bazel to build my test file. 

I used something similar to this:
https://github.com/RobotLocomotion/drake-external-examples/blob/c4b7e210c35a6342d7ad979f5cc166e7edd6f0f4/drake_cmake_external/CMakeLists.txt#L133

```shell
> FAILED: test/Debug/tests test/tests_tests-00.cmake /home/aminya/test/build/test/tests_tests-00.cmake 
: && /usr/bin/g++-11 -g -fsanitize=address,undefined test/CMakeFiles/tests.dir/Debug/tests.cpp.o -o test/Debug/tests  -Wl,-rpath,/home/aminya/test/build/vcpkg_installed/x64-linux/lib  vcpkg_installed/x64-linux/lib/libdrake.so  vcpkg_installed/x64-linux/lib/libdrake_marker.so  vcpkg_installed/x64-linux/lib/libdrake_lcm.so  vcpkg_installed/x64-linux/debug/lib/libspdlogd.a  vcpkg_installed/x64-linux/debug/lib/libfmtd.a  -lpthread  vcpkg_installed/x64-linux/debug/lib/libtinyxml2.a && cd /home/aminya/test/build/test && /home/aminya/cmake/cmake-3.22.0-linux-x86_64/bin/cmake -D TEST_TARGET=tests -D TEST_EXECUTABLE=/home/aminya/test/build/test/Debug/tests -D TEST_EXECUTOR= -D TEST_WORKING_DIR=/home/aminya/test/build/test -D TEST_SPEC= -D TEST_EXTRA_ARGS= -D TEST_PROPERTIES= -D TEST_PREFIX= -D TEST_SUFFIX= -D TEST_LIST=tests_TESTS -D TEST_REPORTER= -D TEST_OUTPUT_DIR= -D TEST_OUTPUT_PREFIX= -D TEST_OUTPUT_SUFFIX= -D CTEST_FILE=/home/aminya/test/build/test/tests_tests-00.cmake -P /home/aminya/test/build/vcpkg_installed/x64-linux/share/catch2/CatchAddTests.cmake

/home/linuxbrew/.linuxbrew/bin/ld: warning: libasan.so.5, needed by vcpkg_installed/x64-linux/lib/libdrake.so, may conflict with libasan.so.6
/home/linuxbrew/.linuxbrew/bin/ld: vcpkg_installed/x64-linux/lib/libdrake.so: undefined reference to `std::unique_ptr<drake::schema::DistributionVector, std::default_delete<drake::schema::DistributionVector> > drake::schema::ToDistributionVector<3>(std::variant<Eigen::Matrix<double, 3, 1, ((Eigen::StorageOptions)0)|((((3)==(1))&&((1)!=(1)))?((Eigen::StorageOptions)1) : ((((1)==(1))&&((3)!=(1)))?((Eigen::StorageOptions)0) : ((Eigen::StorageOptions)0))), 3, 1>, drake::schema::DeterministicVector<3>, drake::schema::GaussianVector<3>, drake::schema::UniformVector<3>, std::conditional<((3)==Eigen::Dynamic)||((3)==(1)), drake::schema::Deterministic, drake::schema::internal::InvalidVariantSelection>::type, std::conditional<((3)==Eigen::Dynamic)||((3)==(1)), drake::schema::Gaussian, drake::schema::internal::InvalidVariantSelection>::type, std::conditional<((3)==Eigen::Dynamic)||((3)==(1)), drake::schema::Uniform, drake::schema::internal::InvalidVariantSelection>::type> const&)'
/home/linuxbrew/.linuxbrew/bin/ld: vcpkg_installed/x64-linux/lib/libdrake.so: undefined reference to `bool drake::schema::IsDeterministic<3>(std::variant<Eigen::Matrix<double, 3, 1, ((Eigen::StorageOptions)0)|((((3)==(1))&&((1)!=(1)))?((Eigen::StorageOptions)1) : ((((1)==(1))&&((3)!=(1)))?((Eigen::StorageOptions)0) : ((Eigen::StorageOptions)0))), 3, 1>, drake::schema::DeterministicVector<3>, drake::schema::GaussianVector<3>, drake::schema::UniformVector<3>, std::conditional<((3)==Eigen::Dynamic)||((3)==(1)), drake::schema::Deterministic, drake::schema::internal::InvalidVariantSelection>::type, std::conditional<((3)==Eigen::Dynamic)||((3)==(1)), drake::schema::Gaussian, drake::schema::internal::InvalidVariantSelection>::type, std::conditional<((3)==Eigen::Dynamic)||((3)==(1)), drake::schema::Uniform, drake::schema::internal::InvalidVariantSelection>::type> const&)'
/home/linuxbrew/.linuxbrew/bin/ld: vcpkg_installed/x64-linux/lib/libdrake.so: undefined reference to `Eigen::Matrix<double, -1, 1, 0, -1, 1> drake::schema::GetDeterministicValue<3>(std::variant<Eigen::Matrix<double, 3, 1, ((Eigen::StorageOptions)0)|((((3)==(1))&&((1)!=(1)))?((Eigen::StorageOptions)1) : ((((1)==(1))&&((3)!=(1)))?((Eigen::StorageOptions)0) : ((Eigen::StorageOptions)0))), 3, 1>, drake::schema::DeterministicVector<3>, drake::schema::GaussianVector<3>, drake::schema::UniformVector<3>, std::conditional<((3)==Eigen::Dynamic)||((3)==(1)), drake::schema::Deterministic, drake::schema::internal::InvalidVariantSelection>::type, std::conditional<((3)==Eigen::Dynamic)||((3)==(1)), drake::schema::Gaussian, drake::schema::internal::InvalidVariantSelection>::type, std::conditional<((3)==Eigen::Dynamic)||((3)==(1)), drake::schema::Uniform, drake::schema::internal::InvalidVariantSelection>::type> const&)'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

jwnimmer-tri, 2022-03-31:
We can't really help with this until we have a reproducible example -- both the source code you're trying to run, and any configuration or build steps required to invoke the error.

The template instantiations in these lines of code _should_ be providing those symbols.

https://github.com/RobotLocomotion/drake/blob/7982ce45c184a7ab215fcb6ba393f0be908d22b8/common/schema/stochastic.cc#L437-L465

jwnimmer-tri, 2022-04-01:
Closing for now, for lack of information to reproduce the problem.

Our issue template points to https://drake.mit.edu/getting_help.html, which says that for debugging help we'd prefer to field questions in StackOverflow.  Only feature requests or reproducible bug reports should be filed as issues.

closed 2022-04-01 by jwnimmer-tri