RobotLocomotion/drake #16911: Meshcat is not working on deepnote

RussTedrake, 2022-04-06:
I hadn't actually updated my coursenotes to use the latest drake binaries since https://github.com/RobotLocomotion/drake/pull/16805 landed.  After updating, launching meshcat on deepnote and opening the printed URL results in 
```
502 Bad Gateway
nginx
```

To reproduce, follow the link to the deepnote tutorials on the drake website, open `rendering_multibody_plant.ipynb`, run the first 3 cells and try to open the meshcat browser.

RussTedrake, 2022-04-06:
reverting to `FROM robotlocomotion/drake:focal-20220315` did not resolve the issue.  Maybe the problem is on deepnote's end?

sammy-tri, 2022-04-06:
I *think* this is a deepnote issue.  In the example notebook, opening a deepnote terminal shows no activity in the logs in `/var/log/nginx`.  I tried creating a fresh deepnote project, enabling incoming connections, and listening on port 8080, and I didn't see any incoming connection attempts when trying to navigate to the URL deepnote provided for my project.

RussTedrake, 2022-04-06:
thank you Sam!

RussTedrake, 2022-04-06:
@sammy-tri -- can you confirm that you set the "Allow incoming connections" flag to true in the environments setting?

sammy-tri, 2022-04-06:
The test configuration shown in my last screenshot worked briefly, as did a meshcat server from a drake tutorial.  Then it went back to 502.

RussTedrake, 2022-04-07:
Deepnote has resolved the issue.  All is well again.
https://community.deepnote.com/c/ask-anything/allow-incoming-connections-is-not-working

closed 2022-04-07 by RussTedrake

lyltc1, 2023-03-10:
I face the same issue now. Meshcat is not working on deepnote [Nginx bad gateway]