RobotLocomotion/drake #16923: proposal: `ConcatenateStdVector : public LeafSystem`

RussTedrake, 2022-04-11:
Pain point:  If I have multiple systems that produces one or more `ExternallyAppliedSpatialForce`, then (I believe) there is no ergonomic way to combine them into a single `std::vector<ExternallyAppliedSpatialForce>` to be passed to the appropriate input port of `MultibodyPlant`.  We have other `std::vector<Type>` input ports in the framework that could similarly benefit from some form of accumulator.

Proposal:  Rather than a narrow implementation for this one type, I think it would be more useful to write a new primitive that looks something like

```
template <typename T, template <typename> typename ValueType>
class ConcatenateStdVector : public LeafSystem<T> {
```
which will offer some number of input ports which accept `std::vector<ValueType<T>>` and some that accept `ValueType<T>`, and a single output port which concatenates them into a single `std::vector<ValueType<T>>`. 

Although I have not tried yet, I believe that writing a method that produces the python bindings for each `ValueType` in much the way we bind e.g. `solvers::Binding`, should be straightforward.

WDYT?

jwnimmer-tri, 2022-04-12:
This might be duplicate of #13139?

In terms of a mux system, my thoughts were https://github.com/RobotLocomotion/drake/issues/12401#issuecomment-617168844.  I still think we should do it type-erased, not with `typename ValueType` statically.

However, stepping back, I don't like the mux system idea much in the first place.

Instead of adding more awkward boilerplate systems that people need to manually wire up with more connection boilerplate, I'd much rather that we add the ability in DiagramBuilder to connect multiple output ports to a single input port, natively.

The developer (in this case, MbP) would need to opt-in input port declaration as allowing multiple connections. Once they do so, then the user can just connect all of the (e.g.) propeller force outputs to the sole MbP input, and call it a day.

RussTedrake, 2022-04-12:
Excellent.  Thank you for finding the old discussion!

I agree that having native support in the system framework would be much better.  But that's a bigger mountain for me to climb this week.  I think that offering the mux does no harm, and it would unblock these workflows right now.  I can write in this idea of the system framework native support into that file.  Sound ok?

jwnimmer-tri, 2022-04-12:
I think a specific ExternalAppliedSpatialForceMultiplexer (in the plant folder) makes a lot of sense as an expedient stopgap.  I think a generic (templated) mux has the problems noted in #13139 and should be avoided.

RussTedrake, 2022-04-12:
Based on the discussion in #13139, I was planning to do the generic, but type-erased version as my stopgap.

EricCousineau-TRI, 2022-10-24:
Started simple version of generic concatenator, but abandoned per Drake Slack thread 7d889f56bb
Going route of `ExternallyAppliedSpatialForceMultiplexer` per #13139

EricCousineau-TRI, 2022-10-26:
Given #18171 landed, should we consider closing this issue until we have needs for multiple types `T` for concatenating `std::vector<T>`?

RussTedrake, 2022-10-27:
Jeremy's idea here about having the vector input ports know how to accept multiple connections seems like an idea worth preserving.  We could rename this issue to track it?

But my original request is satisfied by #18171.

jwnimmer-tri, 2022-10-27:
> Jeremy's idea here about having the vector input ports know how to accept multiple connections seems like an idea worth preserving. We could rename this issue to track it?

=> #18194

closed 2022-10-27 by jwnimmer-tri