RobotLocomotion/drake #16955: Cleanup DiscreteUpdateManager interface

amcastro-tri, 2022-04-13:
As we use the `DiscreteUpdateManager` to place all of our discrete contact dynamics in one place, we realize that the interface provided by managers and how they get used by `MultibodyPlant` is a bit clunky. Currently, in order to talk to an MBP a manager must implement:
1. DoCalcContactSolverResults(), 
2. DoCalcAccelerationKinematicsCache() and,
3. DoCalcDiscreteValues().

`DoCalcDiscreteValues()` makes sense. The manager after all is the one that knows how to perform the discrete update, and `DoCalcDiscreteValues()` is the one entry point to a mutable.

`DoCalcAccelerationKinematicsCache()` is used to report back to MultibodyPlant the discrete accelerations computed by the manger. Since we use a velocity level discrete update to deal with contact, we can always compute the discrete accelerations from the knowledge of the next time step velocities. This is not a problem today, but if we remove it, we simplify the number of required implementations from three to two.

`DoCalcContactSolverResults()` is definitely strange in that it is tied to our old struct to report solver results for TAMSI. For newer solvers supporting more general constraints, `ContactSolverResults` must be updated or replaced completely. The reality however is that all the manger needs to communicate to the plant are `ContactResults`, so that the plant can keep providing ports for visualization and force feedback. That seems to indicate the manger should instead provide something like `EvalContactResults()`. Today, `MultibodyPlant` reconstructs `ContactResults` from `ContactSolverResults`, which not only leads to code duplication but increases the chances of MBP and manager code going out of sync and potentially introducing bugs.
**It is the manager** after all, that knows how to perform this mapping and has access to the data used to perform the actual discrete update.

Therefore this issue proposes to simplify the previous interface to:
1. `DoCalcDiscreteValues()`, as we have it today and,
2. `const ContactResults<T>& EvalMultibodyContactResults(const Context<T>& context) const`, which would report contact results involving multibody elements only (e.g. deformables are not included).

xuchenhan-tri, 2022-04-13:
Thanks for the write up. I like the simplification. Two questions:
1. Will the contact results from continuous simulation continue to be prepared by MbP?
2. Can we extend the `EvalMultibodyContactResults()` function to include deformable contacts as well? The contact results will be similar to those from hydro.

amcastro-tri, 2022-04-13:
1. No. `DiscreteUpdateManager` only takes care of the discrete model. The continuous model will, at least for now, continue to exist in the MultibodyPlant.
2. I imagine that'd require to extend ContactResults to report contact for deformables. Do you think that's possible? Even if it I wonder if maintainable. That is, every time we need to include a new model (say particles or some other contact model). I imagined a world in which additional ports were provided by the manager. But I can be convinced otherwise.

jwnimmer-tri, 2022-04-13:
Having the sole ContactResults output port be essentially a `variant<...>` of all the different contact format we ever support will not scale.  My advice is to have one output port per category of contact, where a manager just populate the one port is knows about (and owns).

For all of the visualization code that I wrote, dealing with `point | hydro` disjunction on that output port was a huge pain in the butt.  I would much rather have had one port per contact type.

Is there any important use case where we need to jam the union of the contact results onto a single output port?

xuchenhan-tri, 2022-04-13:
Having one output port per type of contact results is fine if that makes downstream use cases easier. However, I don't think we should have multiple managers floating around (e.g. one for multibodies only and one for deformable). Eventually, we should have a single manager that handles discrete updates for everything we can simulate.

sherm1, 2022-04-14:
In general, contact forces need to be calculated simultaneously regardless of the model so it likely isn't possible to fully partition them internally. They could be partitioned on output though if that's worthwhile.

amcastro-tri, 2022-04-14:
> However, I don't think we should have multiple managers floating around 

Jeremy suggested different output ports, not multiple managers. I agree with you @xuchenhan-tri that we do want a single manager, but possibly it could provide multiply output ports if needed?

Here is a compromise, instead of `EvalMultibodyContactResults()`, lets have it be `EvalContactResults()` for now. If when deformable lands we do realize their contact results can be reported with exactly the same format, then no new output port is needed.

jwnimmer-tri, 2024-11-01:
@amcastro-tri @xuchenhan-tri Please reply with an updated summary of the state of affairs, and whether or not this is still relevant.  I know the API has already shifted somewhat in the past two years.