RobotLocomotion/drake #16969: Deprecate and remove support for Binder?

jwnimmer-tri, 2022-04-14:
Split out from https://github.com/RobotLocomotion/drake/issues/13683#issuecomment-999807942.

More info (and pro/con discussion) to follow.

jwnimmer-tri, 2022-04-26:
Within Drake, the only mention of "binder" is either in documentation that mentions a long list of potential online notebooks (_"Note that some commercial solvers are not included in the pre-compiled Drake binaries, and therefore not on Deepnote/Colab/Binder."_), or the top-level `.binder` folder.  In drake-ci, we have some [special code](https://github.com/RobotLocomotion/drake-ci/blob/7ab0fedc16645832eb00967a75d001b2bc89a347/driver/configurations/bazel/step-push-nightly-release-branch.cmake#L62-L102) to push a binder-specific branch that pins binder to the most recently nightly [docker image](https://drake.mit.edu/docker.html).

Now that our tutorials are hosted on Deepnote, and  we offer drake wheels on pypi, I think the binder-specific image is not as useful as it used to be.

The only purpose of the dotfile directory in Drake is to run Drake's tutorials on Binder.  I'd rather not support that -- I think we should maintain exactly one tutorials host, so that there's only one thing we need to test.  Thus, we should purge the dotfile directory.

As such, the `drake-ci` logic to update the branch will also need to go.  There no more Dockerfile to re-pin every night.

I don't think we need any deprecation announcement for this.  The binder files and branch are not part of the [Stable API](https://drake.mit.edu/stable.html).

jwnimmer-tri, 2022-04-26:
\CC @RussTedrake I'm going to schedule this for Kitware soon.  Let us know if you have any concerns.

RussTedrake, 2022-04-26:
SGTM

jwnimmer-tri, 2022-05-06:
I've just deleted the `nightly-release-binder` branch (and changed the branch protection rule pattern from `*release*` to just `nightly-release` directly).

PR #17158 (2022-05-11): Remove support for Binder
Resolves #16969

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/17158)
closed 2022-05-11 by BetsyMcPhail