RobotLocomotion/drake #16978: Deprecation of pure-python MeshcatVisualizer

jwnimmer-tri, 2022-04-15:
As it stands, we have two meshcat-based APIs in pydrake, which is a bit confusing for users.

Check the module https://drake.mit.edu/pydrake/pydrake.systems.meshcat_visualizer.html.  There are several classes there, namely `MeshcatVisualizer`, `MeshcatContactVisualizer`, etc.  That's older API, which I think we'd like to deprecate.

The https://drake.mit.edu/pydrake/pydrake.geometry.html#pydrake.geometry.MeshcatVisualizerCpp and related are the new API that we should prefer, instead.

I believe our plan was to finish adding those same features to the C++ code, ensure it's all bound in pydrake, port all of Drake's internal uses away from this module (#14702), and then deprecate it for eventual removal.  

Perhaps we should immediately add a disclaimer atop that module, to dissuade people from using it?

@RussTedrake could you comment whether you agree with the outline above?

Assuming yes, we'll need to make a feature inventory of the old API, and then check off whether the new API is sufficient.  I didn't see a tracking issue for any of this yet, so I made this one.  (If there is already a checklist, we can mark one as a duplicate of the other.)

RussTedrake, 2022-04-18:
Yes.  That's consistent with my original plan: https://github.com/RobotLocomotion/drake/issues/13038#issuecomment-903041628 .  Adding a pre-deprecation note now sounds great.  Or we can move straight to deprecation; I'm not aware of any workflows that would benefit from any further delay.

I would add that my intention was also to rename `MeshcatVisualizerCpp` => `MeshcatVisualizer` at the end of the deprecation dance.

jwnimmer-tri, 2022-05-14:
> ... I'm not aware of any workflows ...

The only thing I see is that you'll need to touch up one of your unit tests:

https://github.com/RussTedrake/manipulation/blob/5e5981147079e69f03d1b42707b8db0386dc8824/manipulation/test/test_model_directives.py#L22-L23

RussTedrake, 2022-05-14:
Should we also schedule the (multi-step) change of `MeshcatVisualizerCpp` => `MeshcatVisualizer` that will be possible when this change lands?

jwnimmer-tri, 2022-05-15:
I think the `MeshcatVisualizer` name collision between the two classes only happens in `pydrake.all`, is that right?  In which case, I should really just roll that rename into the currently-open PR.

RussTedrake, 2022-05-15:
Yes, I think that's accurate... though I've considered it a rule that we don't have the same symbol available in multiple subpackages.  If you change it now, then people who import the meshcat python through `pydrake.all` will not get the deprecation... and will instead get a new class (which might be confusing).  is that right?

jwnimmer-tri, 2022-05-15:
> If you change it now, then people who import the meshcat python through pydrake.all will not get the deprecation... and will instead get a new class (which might be confusing). is that right?

Yes, that's accurate.  That's why `pydrake.all` is documented to be only for temporary hacking, and is not part of the Stable API.

PR #17194 (2022-05-16): [pydrake] Deprecate old Python-based meshcat implementation
Closes #16978.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/17194)
closed 2022-05-16 by rpoyner-tri