RobotLocomotion/drake #16979: Drake Visualize a KUKA arm ERROR: Error reading 'v' at line 3

dvogureckiy99, 2022-04-15:
When I try run this example [Visualize a KUKA arm](https://drake.guzhaoyuan.com/to-get-started)
I have trouble with this command: `./bazel-bin/tools/drake_visualizer`
```
ERROR: In /vtk/src/IO/Import/vtkOBJImporter.cxx, line 495
vtkOBJPolyDataProcessor (0x55b5aa616a20): Error reading 'v' at line 3
```

dvogureckiy99, 2022-04-15:
I resolved this error on my Ubuntu 20.04 system by changing the locale settings.

    1. Run in terminal:
    sudo update-locale LC_NUMERIC=C
    2. Log out and then log in.

jwnimmer-tri, 2022-04-18:
Yes, unfortunately `drake-visualizer` does not correctly ignore the user's locale when parsing `*.obj` files.  Per #16215 though, we don't intent to try to fix that.  The work-around posted above is the correct one.  (Setting the locale just for the visualizer binary is also fine.)

closed 2022-04-18 by jwnimmer-tri