RobotLocomotion/drake #16990: MOSEK loader paths test within Drake

jwnimmer-tri, 2022-04-19:
In #16985, we fixed a bug in Drake's installer that was only ever caught in drake-external-examples CI.  We'd much rather catch this during Drake's normal (Continuous) CI.

We should try to add an `install_test` for this case.

I hope that we could look at the `otool` output for the installed library, and alert on any paths that are incorrectly formulated.

Relatedly, we also have a lot of otool-scraping code currently committed, and more on the way in #16872.  It might also be nice to factor out otool scraping into a Drake-internal python library, so that it can be easily reused everywhere, and so that only one function needs to be carefully commented about the otool output formats, instead of doing brittle string parsing in several places.

\CC @BetsyMcPhail @svenevs FYI

svenevs, 2022-04-20:
See also: https://github.com/RobotLocomotion/drake/pull/16991#pullrequestreview-947476675

When implementing the tests, reverting #16991 and #16985 will be a pathway for "testing the tests."

mwoehlke-kitware, 2022-06-14:
Can we not add a test that the installed libraries are _usable_? That seems like it would be more valuable than adding ad-hoc tests for individual libraries.

jwnimmer-tri, 2022-06-14:
The problem comes when the paths refer to libraries that exist, but only in the build tree.  It's difficult to make the build tree evaporate or become inaccessible when running the tests.  That's why an allow-list for well-formed paths is probably the most cost-effective testing mechanism.

PR #17858 (2022-09-08): Add install test to verify library paths
Add an install test to inspect the set of libraries which libdrake.so consumes, specifically looking for any that are outside the install tree or system paths.

Closes #16990.

- - -
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/17858)
PR #17902 (2022-09-14): Add install test to verify library paths
Add an install test to inspect the set of libraries which `libdrake.so` consumes, specifically looking for any that are outside the install tree or system paths.

Closes #16990.

This is a re-application (un-revert) of #17858 after fixing post-merge CI problems.

- - -
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/17902)
closed 2022-09-14 by jwnimmer-tri