RobotLocomotion/drake #16994: LeafSystem::DeclareCacheEntry doxygen missing args.

ggould-tri, 2022-04-19:
See https://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1_system.html#a7da4e8aa9a414faa5b40fc64e8ca40e3
[image]
The version of `LeafSystem::DeclareCacheEntry` that takes a `value_producer` appears as having no arguments, even though it has three arguments.

xuchenhan-tri, 2022-04-19:
Seems like a systematic issue with `using` base class functions. For example, it also happens here: https://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1_context.html#a415588e5494582fba4c5504ccc2ae795

sherm1, 2022-04-19:
Looks like a Doxygen issue -- file [here](https://github.com/doxygen/doxygen/issues)?

jwnimmer-tri, 2022-04-19:
Our Doxygen binary is purposefully pinned to less-that-latest, because the latest releases of Doxygen have _even more_ bugs.  Before reporting this upstream, be sure to reproduce it using the latest release of upstream Doxygen.

That being said, as far as the Drake issue goes we could also try to find a work-around.  For example, the [SystemBase](https://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1_system_base.html#a7da4e8aa9a414faa5b40fc64e8ca40e3) documentation is correct.  We could exclude the `using` statement within `class System` from Doxygen (via the `DRAKE_DOXYGEN_CXX` macro) to remove the malformed content from the API reference.

xuchenhan-tri, 2022-04-19:
I like the proposed solution by @jwnimmer-tri. The `using` statement is there for qol/convennience. The additional doxygen entry doesn't add much value.

ggould-tri, 2022-04-19:
The cure sounds worse than the disease here.  Having to traverse down through `LeafSystem` / `System` / `SystemBase` to find methods -- or answer "is there a method for this" -- is definitely not an improvement!

jwnimmer-tri, 2022-04-19:
Just to be clear, the original issue post was inaccurate.  It claimed that "LeafSystem" was missing the named arguments; in fact, the hyperlink was to the `class System` documentation.

The `class LeafSystem` documentation _does_ show `DeclareCacheEntry()` correctly.  You need to unfold the triangle image "Protected member functions inherited from SystemBase" in order to see it.  The same is actually true for the `class System` documentation.  The arguments are _only_ missing from the documentation induced by the `using` statement.

ggould-tri, 2022-04-19:
Huh, I guess copy-pasted the wrong class name from my innumerable doxygen tabs.
Anyway I'm going to close this issue and flee the field before things get any worse.

closed 2022-04-19 by ggould-tri

jwnimmer-tri, 2022-04-19:
It still seems like an acute issue (missing args) with a trivial fix (adding a cross-reference hyperlink), but in any case since I'm not the one volunteering to fix it I'll let @sherm1 or @xuchenhan-tri decide if they want to re-open it.

sherm1, 2022-04-19:
I filed Doxygen issue [#9282](https://github.com/doxygen/doxygen/issues/9282).  Don't hold your breath, though -- there are 1900 open issues there!

jwnimmer-tri, 2022-04-19:
For the record, as of today we're using Doxygen 1.8.15

https://github.com/RobotLocomotion/drake/blob/e58be44e3fa042ee796c49de61c1d9dbf9bbf441/tools/workspace/doxygen/Dockerfile#L25

sherm1, 2022-04-19:
Thanks -- I added that to the Doxygen issue.

sherm1, 2022-04-20:
Per a Doxygen developer, the missing-parameters problem is already fixed in current Doxygen 1.9.3. Proof [here](https://github.com/doxygen/doxygen/issues/9282#issuecomment-1103619349).