RobotLocomotion/drake #17022: `This method doesn't support T = drake::symbolic::Expression.` without callstack in Deepnote

RussTedrake, 2022-04-23:
We have a number of contact-related methods in `MultibodyPlant` that throw either a `std::logic_error` or `std::domain_error` with a message of the form `This method doesn't support T = drake::symbolic::Expression.`

Problem: In python (or at very least Deepnote), the throwing call stack is not shown, so the user **only** sees the message
```
ValueError: This method doesn't support T = drake::symbolic::Expression.
```
This is very confusing, because they believe that the calling method is throwing, or they make a small chance to their URDF and suddenly get this message and simply don't know why.

Expected result:
The user gets a clearer message.  

Am I right that the message is more clear in c++, since the method name gets shown in the call stack?  If yes, then could we find a way for that level of message reporting appears in python/Deepnote?

If no, can we at least make these 6 messages at least give a more helpful error, for instance by naming the function that is calling them?

@rpoyner-tri

jwnimmer-tri, 2022-04-23:
> Am I right that the message is more clear in c++, since the method name gets shown in the call stack?

For our C++ library / programs generally, we do not print a call stack for exceptions.  The user would need to fire up `gdb` to inspect the call stack of an exception.  The error message friction here probably applies equally well to C++ and Python.

rpoyner-tri, 2022-04-25:
For extra fun, none the methods that throw "This method doesn't" exceptions are public, so the immediate function names aren't much use.

rpoyner-tri, 2022-04-25:
Tracing the causes and effects by hand, it looks to me like all of the "This method doesn't" culprits are pieces of hydroelastic contact query, and only throw when the scalar type is symbolic::Expression. Since the actual methods are (a) private, and (b) often invoked via the cache system, their identities as functions are fairly useless clues for debugging.

Instead, I'm thinking of something like the list below. These are not alternatives; the solution would implement all of the listed items.

* On creation of a plant with T = Expression, warn if it comes into existence with contact_model that invokes hydroelastic queries.
* On `set_contact_model()` of a T=Expression plant, warn if the new mode is a hydro mode.
* On Finalize() of a plant with T=Expression and hydro contact, throw with a verbose explanatory error message.
*  Leave the existing throws in place, but improve the messages.

I'm open to alternatives, but probably not "implement symbolic hydro" -- at least, not within the scope of this issue.

jwnimmer-tri, 2022-04-25:
For options 1-3, in case a user wants to use `MbP<Expression>` but is not going to call any functions related to contact (nor its associated dynamics, etc), we must not throw an exception.  For example, the user can still do kinematics, in this case.  We also should not _warn_ in this case, since it will just be console spam.  Since hydroelastic is the _default_ now, I think there would be a lot of false positives.

> Leave the existing throws in place, but improve the messages.

This was my initial thought when I skimmed the code.  Saying "the hydroelastic contact model does not support symbolic expressions" is probably good enough?  I _think_ it's the case that the user would never see this messages, unless they have annotated their models (e.g., URDF files) with hydroelastic properties?

RussTedrake, 2022-04-26:
Just to be clear... there are a few call sites that are not related to hydroelastic.  The students that ran into this were not enabling hydro when they hit this error.  I see `CallContactSolver`, `CalcDiscreteContactPairs` which, I believe, are not exclusively hydro.

jwnimmer-tri, 2022-04-26:
The `CallContactSolver` would not be involved; that is a vestigial internal API that does not appear to be used anywhere.

The `CalcDiscreteContactPairs` (as used forward dynamics, I believe) seems more likely.  Perhaps rico has a handle on that.

rpoyner-tri, 2022-04-26:
I agree that my first proposal was probably too extreme and disables some useful calculations. I think a confounding factor is that the default `contact_model` changed to hydroelastic-with-contact recently. So, likely we have a lot of unsuspecting users waltzing into the hole created by "no symbolic hydro" without having made any obvious/explicit choices.

I'll do some more study and see. I fear that `CalcDiscreteContactPairs<Expression>` will always throw in either hydro or hydro-with-fallback modes.

rpoyner-tri, 2022-04-26:
It would help me, as a non-expert Drake user, to have some plausible scenarios in hand to reproduce the problem actually seen. Is there a notebook or python program around that provokes the symptom?

RussTedrake, 2022-04-26:
This is the report that I had:

I'm playing with the code to extract the lumped parameters for the cartpole system (Ch. 18 notebook), except by applying it to my system with an iiwa and an object attached with a pin joint. I'm trying to run the function
```
MultibodyPlant_[Expression].CalcImplicitTimeDerivativesResidual(...)
```
but see this error (that does not happen when I run the notebook):
```
> residual = sym_plant.CalcImplicitTimeDerivativesResidual(
ValueError: This method doesn't support T = drake::symbolic::Expression.
```
The notebook would lead me to believe that it does support this type, since all of the types of my plant and parameters are the same. Thank you for your help!

jwnimmer-tri, 2022-04-26:
@rpoyner-tri:

> I think a confounding factor is that the default contact_model changed to hydroelastic-with-contact recently. So, likely we have a lot of unsuspecting users waltzing into the hole created by "no symbolic hydro" without having made any obvious/explicit choices.

That's my read as well.  The forward dynamics involves contact forces, and when in "hydro with fallback" mode we throw an exception when trying to add in those forces from any symbolic contact surfaces (even though there would be no such forces in this example).

That's means another work-around for users (instead of deleting their collision geometries) would be to switch the plant's contact model to point contact only, instead of fallback (see `MultibodyPlant::set_contact_model`).

I think the problem stems from throwing the exception during MbP's internal cache entry evaluation.  That's the wrong layer to implement this check.  It's the QueryObject that should throw an exception, not the plant, since the QueryObject is the place that knows whether there are any such contacts.  And when the QueryObject is asked to compute the answer for Expression, it can return "no contacts" when no hydroelastic geometries have been tagged, or throw when there are some.

rpoyner-tri, 2022-04-26:
@ggould-tri floated the idea of having symbolic contact return symbolic nan, or perhaps function-wrapped assert or throw. This might have the nice property that we could just remove all the specializations from mbp, and have the excuses live in the geometry layers.

jwnimmer-tri, 2022-04-26:
I am not sure what nan values would buy us?  My intuition is that immediately throwing iff any surface contact is possible should suffice.  Deferring the failure until the values are being process with a "NaN detected" error message is likely unhelpful.

(I do support moving the "I can't handle T=Expression" logic into geometry, instead of the plant.)

rpoyner-tri, 2022-04-27:
Leaving this for now, to prioritize #adoption work.

rpoyner-tri, 2022-04-27:
@joemasterjohn would you be interested in taking a look at this? It is (in part) a consequence of making hydro the default model.

jwnimmer-tri, 2022-04-27:
To be clear, I think there are two distinct problems here:

(1) When we report "This method doesn't support T = drake::symbolic::Expression", there is a usability issue to help the user understand what they did wrong and how to resolve the problem.  The current message doesn't do much to assist.

(2) It seems probable that #16569 introduced a significant regression in MbP, where users with geometries that use point contact could do so with a symbolic MbP as long as there were never any collisions in practice.  Now, they get an exception when trying to do anything related to MbP dynamics.

To me, the priority of (1) is not very severe (we could sort it into the MbP usability backlog and return to it later), but (2) represents a major bug that escaped our testing during the hydro release process, and thus should be attacked with with higher priority than adding new features (SAP, FEM, etc).

sherm1, 2022-04-28:
Leaving this assigned to Rico for now until we find a home for it.

amcastro-tri, 2022-04-28:
Per f2f with @rpoyner-tri:

1. I believe any `SceneGraph::CalcContactQuery<symbolic::Expression>` throws, or should throw.
2. The fact that previous default "point" worked, most likely was caused by a "shortcut". With "shortcut" I mean most likely an `if` statement that is not properly converted to `cond`.
3. Even if (2) is itself a bug, the proper solution could be so overly complex to implement in practice that this bug might be the best line of action right now (probably with additional warning and/or throw messages to the user, whenever applicable/possible)

So I think what we should try here is:
1. Understand the data flow in the "point" contact mode only and understand why we get the "desired" behavior (at least desired in the sense Russ describes above for his lumped parameters application)
2. Replicate this data flow when using hydroelastic contact (easier said than done, but probably not crazy if we attack (1)?).

DamrongGuoy, 2022-04-28:
I like _"create a reproduction recipe (or unit test) that probes this behavior, and running under the two contact modes (point, fallback)."_  If we can make it into unit tests, we can use them for code maintenance.

DamrongGuoy, 2022-04-28:
Are these details relevant?

On the `geometry` side, QueryObject's doxygen has tables like this one:
[image]
for [ComputePointPairPenetration](https://drake.mit.edu/doxygen_cxx/classdrake_1_1geometry_1_1_query_object.html#af3b16adde799eb804619dc86f3703f4f) (), [ComputeSignedDistancePairwiseClosestPoints](https://drake.mit.edu/doxygen_cxx/classdrake_1_1geometry_1_1_query_object.html#a0918d61334404a7680145cb882a18d52)().  
It also has this table for [ComputeSignedDistanceToPoint](https://drake.mit.edu/doxygen_cxx/classdrake_1_1geometry_1_1_query_object.html#aa4c7754ca66a8593b182ae990643875c)():
[image]

Unfortunately, we don't have a similar table for other functions like [ComputeContactSurfaces](https://drake.mit.edu/doxygen_cxx/classdrake_1_1geometry_1_1_query_object.html#ab15e2fff4faa847eb6a7ba82e1d4f035)() and  [ComputeContactSurfacesWithFallback](https://drake.mit.edu/doxygen_cxx/classdrake_1_1geometry_1_1_query_object.html#a9c5e3da8410ee9aa2ae555912d9c98e6)(), which is for hydroelastic contact with fallback to point contact.

rpoyner-tri, 2022-04-28:
I have a repro branch, which gets to the original error message, but maybe isn't quite right yet: https://github.com/rpoyner-tri/drake/tree/this-method

Repro in hydro-with-fallback:

```
rico@puget-161804-10:~/checkout/drake$ bazel run -c dbg //examples/hydroelastic/python_ball_paddle:contact_sim_demo_py -- --time_step 0
INFO: Analyzed target //examples/hydroelastic/python_ball_paddle:contact_sim_demo_py (0 packages loaded, 0 targets configured).
INFO: Found 1 target...
Target //examples/hydroelastic/python_ball_paddle:contact_sim_demo_py up-to-date:
  bazel-bin/examples/hydroelastic/python_ball_paddle/contact_sim_demo_py
INFO: Elapsed time: 5.730s, Critical Path: 4.89s
INFO: 39 processes: 10 internal, 29 linux-sandbox.
INFO: Build completed successfully, 39 total actions
INFO: Build completed successfully, 39 total actions
Traceback (most recent call last):
  File "/home/rico/.cache/bazel/_bazel_rico/7f8997f28c9253517a55d673c67a6c74/execroot/drake/bazel-out/k8-dbg/bin/examples/hydroelastic/python_ball_paddle/contact_sim_demo_py.runfiles/drake/examples/hydroelastic/python_ball_paddle/contact_sim_demo.py", line 153, in <module>
    diagram, ball_paddle_plant, state_logger = make_ball_paddle(
  File "/home/rico/.cache/bazel/_bazel_rico/7f8997f28c9253517a55d673c67a6c74/execroot/drake/bazel-out/k8-dbg/bin/examples/hydroelastic/python_ball_paddle/contact_sim_demo_py.runfiles/drake/examples/hydroelastic/python_ball_paddle/contact_sim_demo.py", line 89, in make_ball_paddle
    symbolic_nonsense(diagram, plant)
  File "/home/rico/.cache/bazel/_bazel_rico/7f8997f28c9253517a55d673c67a6c74/execroot/drake/bazel-out/k8-dbg/bin/examples/hydroelastic/python_ball_paddle/contact_sim_demo_py.runfiles/drake/examples/hydroelastic/python_ball_paddle/contact_sim_demo.py", line 39, in symbolic_nonsense
    res = px.CalcImplicitTimeDerivativesResidual(cx, derivs)
RuntimeError: CalcHydroelasticWithFallback doesn't support scalar type T = symbolic::Expression.
rico@puget-161804-10:~/checkout/drake$ 
```
Unfortunately point contact seems to hit some maybe irrelevant failure. I suspect I've set up the systems wrong somehow.

```
rico@puget-161804-10:~/checkout/drake$ bazel run -c dbg //examples/hydroelastic/python_ball_paddle:contact_sim_demo_py -- --contact_model point --time_step 0
INFO: Analyzed target //examples/hydroelastic/python_ball_paddle:contact_sim_demo_py (0 packages loaded, 0 targets configured).
INFO: Found 1 target...
Target //examples/hydroelastic/python_ball_paddle:contact_sim_demo_py up-to-date:
  bazel-bin/examples/hydroelastic/python_ball_paddle/contact_sim_demo_py
INFO: Elapsed time: 0.706s, Critical Path: 0.00s
INFO: 1 process: 1 internal.
INFO: Build completed successfully, 1 total action
INFO: Build completed successfully, 1 total action
Traceback (most recent call last):
  File "/home/rico/.cache/bazel/_bazel_rico/7f8997f28c9253517a55d673c67a6c74/execroot/drake/bazel-out/k8-dbg/bin/examples/hydroelastic/python_ball_paddle/contact_sim_demo_py.runfiles/drake/examples/hydroelastic/python_ball_paddle/contact_sim_demo.py", line 153, in <module>
    diagram, ball_paddle_plant, state_logger = make_ball_paddle(
  File "/home/rico/.cache/bazel/_bazel_rico/7f8997f28c9253517a55d673c67a6c74/execroot/drake/bazel-out/k8-dbg/bin/examples/hydroelastic/python_ball_paddle/contact_sim_demo_py.runfiles/drake/examples/hydroelastic/python_ball_paddle/contact_sim_demo.py", line 89, in make_ball_paddle
    symbolic_nonsense(diagram, plant)
  File "/home/rico/.cache/bazel/_bazel_rico/7f8997f28c9253517a55d673c67a6c74/execroot/drake/bazel-out/k8-dbg/bin/examples/hydroelastic/python_ball_paddle/contact_sim_demo_py.runfiles/drake/examples/hydroelastic/python_ball_paddle/contact_sim_demo.py", line 39, in symbolic_nonsense
    res = px.CalcImplicitTimeDerivativesResidual(cx, derivs)
RuntimeError: The geometry query input port (see MultibodyPlant::get_geometry_query_input_port()) of this MultibodyPlant is not connected. Please connect thegeometry query output port of a SceneGraph object (see SceneGraph::get_query_output_port()) to this plants input port in a Diagram.
rico@puget-161804-10:~/checkout/drake$ 
```

rpoyner-tri, 2026-11-12:
Reassigning to @joemasterjohn, the `multibody plant` lead per the component table in GitHub Issue Management. I'm not actively working on this.

A note on scope: Deepnote is no longer supported, but it was incidental here. Drake doesn't print call stacks for C++ exceptions either, so Python users see only the bare message. The "This method doesn't support T = Expression" throws are still on master (`multibody_plant.cc` and `discrete_update_manager.cc`), and the message still doesn't say what the user did wrong. The thread also found a related behavior question: with hydroelastic as the default, a symbolic plant that has collision geometry can throw where point contact once worked. I haven't re-checked whether that still reproduces.

@joemasterjohn, please keep it, relabel it, or close it as you see fit.