RobotLocomotion/drake #17023: Solver options `kPrintToConsole` does not cause any output to be visible on Deepnote

RussTedrake, 2022-04-23:
To reproduce: run the `debug_mathematical_program` notebook on Deepnote (as linked from the Drake page).
Expected result: I expect to see solver output after the second cell of the `Enable solver verbosity` section.  Currently I see nothing.

My understanding, based on student questions, is that this is a problem for other solvers as well.

I expect that using `kPrintToFile` is a reasonable work-around until this issue is fixed.  But I do think our tutorial as written, hosted on Deepnote, is not working as advertised.

jwnimmer-tri, 2022-04-23:
See also the #15342 discussion more broadly and the https://github.com/RobotLocomotion/drake/issues/15342#issuecomment-876490253 specifically.

The easy improvement here is to update the text of the tutorial to document that "console" is typically useless in notebooks, and suggested work-around (print to a file, and then read it back).

jwnimmer-tri, 2022-04-25:
I'll work on fixing the tutorial to perform correctly online.  I have a branch started at https://github.com/jwnimmer-tri/drake/tree/tutorials-print-to-file.  The current problem is that `IpoptSolver` does not support `kPrintFileName`, so I'll need to add that feature before I can demonstrate it in the tutorial.

PR #17035 (2022-04-27): [tutorials] Avoid kPrintToConsole in online notebooks
Requires:
- [x] #17034

Closes #17023.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/17035)
closed 2022-04-27 by sherm1