RobotLocomotion/drake #17052: parsing, diagnostics: Can XPath be emitted?

EricCousineau-TRI, 2022-04-27:
Relates #16784

When dealing with xacros / other forms of XML generation, sometimes it's nice to see the XPath

Is it possible to emit XPath (or add it as an option in Parser diagnostics) for URDF + SDFormat, eventually MJCF?
Example of getting XPath from SDFormat: https://github.com/ignitionrobotics/sdformat/pull/533
(glancing at tinyxml2 docs, I don't see explicit XPath, but I think it wouldn't be too hard to make for URDF?)

@rpoyner-tri Assigning to ya for disposition (\cc @sammy-tri @jwnimmer-tri)
If y'all think this is a worthy objective, then you can assign it back to me and I can work w/ OSRC to queue this up.

jwnimmer-tri, 2022-04-27:
For localizing an error condition, I think that a filename and line number mostly serves the need.  Possibly a column number would also help, and/or echoing back the faulty line with a carot (as compilers sometimes do)?

We already report some of that data some of the time, but to the extent that Drake doesn't yet succeed at reporting those details for all parsing errors, I would **definitely** appreciate help it adding it.

As far as XPath goes, my own impression is that to a first approximation nobody understands its syntax.  (Some experts do, but they are probably not the people who need help with parsing diagnostic feedback.)

I'd propose approaching the question in the opposite direction.  Rather than proposing a solution (xpath) in search of a problem (???), can we identify acute deficiencies of the current feedback channel (perhaps also imagining that missing line number features have been fully implemented), and then think about how to solve the remaining challenges?

EricCousineau-TRI, 2022-09-22:
In reviewing MbP issues, came back to this.
Yup, those're all good points, #17053 definitely states a better need, and I have not yet come across a more specific deficiency for this. Closing!

closed 2022-09-22 by EricCousineau-TRI