RobotLocomotion/drake #17061: Lint model files

jwnimmer-tri, 2022-04-28:
As discussed in #15018 and especially https://github.com/RobotLocomotion/drake/issues/15018#issuecomment-1111449656.

We should have a linter for our model files (SDFormat, URDF, etc.) that will report test failures in case any of the model file(s) emit warnings or errors when loaded into MultibodyPlant + SceneGraph.

Broadly, we'd want to lint all of the installed model files plus any other example or tutorials model files.  We should also _probably_ lint unit-test-only model files, but at least the `multibody/parsing` unit test files are _expected_ to have errors, so we'd need to tailor that.

It's probably worth doing #8271 at the same time.  We can check the file with an XML linter first, and then the Parser second.

Also tangentially related is #10610, though that is more about style and clarity than anything acute.  Possibly the linter tool here could grow into that role, in time.

\CC @EricCousineau-TRI @rpoyner-tri FYI

I seem to recall that @rpoyner-tri already had a command-line program that ran the Parser on an input file, but I don't immediately see it.

EricCousineau-TRI, 2022-04-28:
If it's helpful, see also Anzu PR 8773 (mayhaps redundant w.r.t tools in Drake)

rpoyner-tri, 2022-04-28:
multibody/parser/test/parser_manual_test is what you are looking for. It currently doesn't have a great workflow for working with models that need to be bazel-built;  the result is a lot of hard errors from ResolveUri.

There is a model-tester test among the pydrake tests that might be easier to grow into what you want.

jwnimmer-tri, 2022-11-09:
This has fallen below my TODO list cutoff.  If you like @rpoyner-tri feel free to steal it from me.