RobotLocomotion/drake #17068: Make default dummy variable more obvious

richardrl, 2022-04-29:
I was perplexed by this error for a while:
`(Pdb) Evaluate(cost_fnx.Jacobian(x_var), dic)
*** RuntimeError: The following environment does not have an entry for the variable x(0)
x_var(1) -> 2
x_var(0) -> 2
`

The reason is, in my code, my "x_var" variable was named "x":
`    x_var = prog.NewContinuousVariables(x_init.shape[0], "x")
`

But the reason the error was happening was because there was another _unnamed_ variable that did not have an entry in `dic`, but because the Drake default variable name for unnamed variables is `x`, I could not see this.

I would suggest making the default variable name `DummyVariable` or `UnnamedVariable` to avoid this issue.

jwnimmer-tri, 2022-04-29:
This doesn't align with my understanding of symbolic variables and mathematical program. I don't think there's any concept of an "unnamed" variable.  Could you please provide a reproducible example that can illustrate the problem?

hongkai-dai, 2022-04-29:
I think @richardrl means this
https://github.com/RobotLocomotion/drake/blob/e440c8f88f5b8496865339090e2178eead5b5e5f/solvers/mathematical_program.h#L210-L211
that the variables by default have name "x".

He suggests to change that default name to "unnamed".

jwnimmer-tri, 2022-04-29:
No doubt the error message was insufficient to help diagnose the problem.  I would still like to see a fully-worked example, so that we can explore the entire space of mitigations.  My intuition says that playing games with variable names is not going to be the best improvement that helps users understand their situation.

hongkai-dai, 2022-05-04:
@jwnimmer-tri I think if we set the default variable name to "unnamed", then we can run into this issue
```py
x = prog.NewContinuousVariables(2)
y = prog.NewContinuousVariables(2)
```
then both `x` and `y` have the name "unnamed", they are still indistinguishable from each other.

How about we do this
1. If the user doesn't not provide a name, then we name the variable as `unnamed + variable_index`. 
2. If the use does provide a name then we just use that name + count of variable in NewContinuousVariable, and it is the user's responsibility to pass in a different name for each call of NewContinuousVariable

So if we have the following code
```py
x = prog.NewContinuousVariables(2, "x")  # x has name ["x0", "x1"]
y_unnamed = prog.NewContinousVariables(2)  # y has name ["unnamed2", "unnamed3"]
z_unnamed = prog.NewContinousVariables(2)  # z has name ["unnamed4", "unnamed5"]
w = prog.NewContinousVariables(2, "w") # w has name ["w0", "w1"]
```
But if the users call `NewContinousVariables` for twice and pass the same name string, then the variables will share the same name
```py
x1 = prog.NewContinousVariables(2, "x")  # x1 has name ["x0", "x1"]
x2 = prog.NewContinousVariables(2, "x")  # x2 also has name ["x0", "x1"]
```

What do you think?

jwnimmer-tri, 2022-05-04:
I think might I understand the bug report better now.  I think the original description is a bit misleading, but I think what you're getting at now is this kind of error message:

```
The following environment does not have an entry for the variable x(0):
  x(0) -> 10.0
  x(1) -> 11.0
```

We _do_ have `x(0)` provided in the environment, but the problem is that there are two distinct variables that each have the same name (but different IDs).  In MathematicalProgram we reject duplicate variable IDs, but we allow duplicate variable names.

I still don't understand a concrete example of when this would happen, though.  I really do think we need a fully-demonstrated example where this failure was unclear, before we start trying to design solutions.

For example, possibly we should alert when duplicated variable names are introduced (no matter where they came from), rather than worrying about default names.

---

On the topic of that the "new variables" function semantics, though:

I like the default of using `x` for decision variables.  I lot of times we print them out and there are many of them, so compactness is important.  I think probably "x" is also the conventional name for decision variables in the literature?  It seems like a good default value.

Here's another idea:

```c++
/// ...
///
/// If name is provided, then the variables will be named like `name(#)` where `#`
/// counts up from zero.
///
/// If name is not provided, then the variables will be named like `x(#)` where `#` is
/// the decision variable index as per `FindDecisionVariableIndex`.
VectorXDecisionVariable NewContinuousVariables(
    int rows, const std::optional<std::string>& name = std::nullopt);
```

We could also do that _plus_ use a different token for the default.  Possibly some unicode `x` that looks different than the ASCII `x`, to indicate that the index is absolute within the program, rather than within just one constraint?

hongkai-dai, 2022-05-04:
> I like the default of using x for decision variables. I lot of times we print them out and there are many of them, so compactness is important. I think probably "x" is also the conventional name for decision variables in the literature? It seems like a good default value.

I agree that compactness is important. 

On the other hand, a lot of times people call `NewContinuousVariables(num_vars, "x")`, that the user wants to name the variable as `x`. Then in the following example, the two variables `x_unnamed` and `x_named` will be indistinguishable
```py
x_unnamed = prog.NewContinuousVariables(2)
x_named = prog.NewContinuousVariables(2, "x")
```

I like the idea of adding a different token for the default. So how about using token "$"? Then the default name of the variable is "$x(#)" where # is the result of FindDecisionVariableIndex. I think this will distinguish from the variables named "x(#)". What do you think?

jwnimmer-tri, 2022-05-04:
Is `$x(0)` vs `x(0)` any less confusing?  Maybe epsilon, but I'd argue not very much.

It still seems to me like the second line should just throw an error, i.e., when the user adds duplicated variable names into the program.  That, combined with my `nullopt` proposal allows for either of these two workflows:

Use `NewContinuousVariables(N)` with no names, ever.  All variables are created using decision variable indices on `x(#)`, and are therefore automatically unique.

Use `NewContinuousVariables(N, name)` but always with your own, user-provided meaningful name prefix.  If you ever re-use a name, it's an error.  You can still ask `NewContinuousVariables(N)` with no name, but in that case you'd better not use `x` as the a user-specified name.

In other words, I think the problem here is that the user is using `"x"` as a meaningful variable name.  I think we should discourage that, instead relying on the default in that case.

hongkai-dai, 2022-05-04:
I see, I like this proposal.

I think there is probably one caveat. We also have the function `prog.AddDecisionVariables(vars)`, where vars is created somewhere else. If we do not allow duplicate names, then `AddDecisionVariables(vars)` will also throw an error when `vars` has variables already contained in prog. This could be a problem since that `vars` might be created somewhere else by a different user, and the user of this `prog` cannot modify the variable name (once a variable is created, we cannot modify its name).

jwnimmer-tri, 2022-05-09:
> ... If we do not allow duplicate names, then `prog.AddDecisionVariables(vars)` will also throw an error when vars has variables whose names are already contained in `prog`...

How much would this be a problem in practice?

Perhaps we can draft up a "fail on duplicate names" patch and then see what Drake CI (and Anzu CI) has to say about the prevalence of duplicates?

We could offer a deprecation period for this new error-checking, to allow time for downstream code to adjust.  Possibly we could also have a `prog.set_duplicate_names_allowed(true)` opt-out for users who want to use duplicate names, and live with confusing outputs anyway.

closed 2025-08-24 by jwnimmer-tri